ovro-lwa/lwa-issues #7: Summarize and compare all flaggers. Final Stokes V noise in 2-hour dataset.

marinanderson, 2020-10-02:
Compare noise in Stokes V, 2-hour integration, subband 04 using
1. Default pipeline
2. AOFlagger (Bharat's "less sensitive" and "sensitive" strategies)
3. m-mode pipeline (default threshold and Phil's lower threshold for more aggressive flagging)

marinanderson, 2020-10-02:
From original implementation of pipeline (original calibration solutions, no automated antenna flagging from post-cal autocorrs):

1. Default pipeline: 149 mJy
2. AOFlagger:
    - Less sensitive: 145 mJy
    - Sensitive: 143 mJy
3. m-mode:
    - standard: 146 mJy
    - more aggressive: 146 mJy

marinanderson, 2020-10-02:
From latest implementation of pipeline (new calibration solutions, includes automated antenna flagging from post-cal autocorrs):

1. Default pipeline: 124 mJy
2. AOFlagger:
    - Less sensitive: 118 mJy
    - Sensitive: 118 mJy
3. m-mode:
    - standard: 127 mJy
    - more aggressive: 116 mJy

marinanderson, 2020-10-02:
Note: flags from AOFlagger and m-mode pipeline were evaluated by merging with the default pipeline flags (e.g. with an `or` statement).

yupinghuang, 2020-10-02:
I wonder if it's better to do a test without merging the flags. A flagging strategy that does not quite work may still flag some visibility by chance that would improve the sensitivity by a little.

marinanderson, 2020-10-16:
Updating with results from SSINS:

Latest implementation of pipeline (new calibration solutions, includes automated antenna flagging): 124 mJy