ovro-lwa/lwa-issues #14: raw method of python client ARX class does not return what is expected.

rh-codebase, 2021-06-02:
via e-mail from @ldaddario :                                                                                                      
> > I expected it to display the exact response string that it got from the                                                                                                          
> > bus (typically but not necessarily hex numbers), but instead it gives                                                                                                          
> > the character code of each character as a 2-digit decimal number, with                                                                                                         
> > all of them run together.  So the display length is twice as many                                                                                                              
> > characters as were received and it's unreadable.  And it does not seem                                                                                                         
> > to include the ACK|NAK character.  I want to see exactly the characters                                                                                                        
> > received from the bus, nothing else.

rh-codebase, 2021-06-02:
The raw() command belonging to the python client ARX API displays the exact bytes read from the bus in hex. Is the expectation to display the ascii representation of the response? Since not all ascii chars are printable, displaying the hex values is more useful in cases where there may be corruption on the bus or the ARX firmware has a bug and does not send back printable ascii. An example:                                    
                                                                                                                                                                                   
>>> ma.raw(2,'ECHOabc')                                                                                                                                                            
'4543484f414243'                                                                                                                                                                   
                                                                                                                                                                                   
The input is 7 ascii chars so hence 7 bytes. The return is 9 bytes. It passes the protocol definition of a proper message so no exception is throw. The bytes in hex: 0x45 = E, 0x43 = C, etc. are displayed. It is exactly what is read from the serial stream with the 'ACK'  and termination '0x0d' stripped as all read streams are sent through a check method to determine if it's a correctly formed message and to decide if an exception should be thrown. All messages being read off the serial bus are checked. Since there's no branch point, there's no way to circumvent the code path.  If an exception is not thrown, then it's guaranteed to be a message starting with 'ACK' and ending with '0x0d'.                                                                 
                                                                                                                                                                                   
If there's a desire to	output an ascii string,  rawascii() can be added to display the output as such.

ldaddario, 2021-06-02:
> The raw() command belonging to the python client ARX API displays the
> exact bytes read from the bus in hex.

raw() is a function, not a command.  It does not "display" anything but
instead returns a value.  That value must be exactly the string of
characters received on the RS485 bus in response to a command.  You took
the actual reply string and converted it to a string twice as long.
Please do no conversions whatsoever.

If the return value from this function (or any function) is to be
displayed, it's up to the caller to do that.

...
> If there's a desire to output an ascii string, rawascii() can be added
> to display the output as such.

No.  There is no need for an additional function.  The current behavior
of raw() is useless and there is no need to keep it.  Please make it
behave correctly.

The current raw() is wrong in the opposite direction too, and that must
also be fixed.  Exactly the string of characters in its 'cmd' argument
must be sent on the RS485 bus without any conversions whatsoever.  The
current version converts lower-case letters to upper case.

The whole idea here is to provide *transparent* access to the RS485 bus.
 If you try to "fix" what the caller wanted to send, or if you modify
what was received, you have defeated the fundamental purpose.

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
        +1/626/510-6492 (home)
        +1/626/657-6357 (mobile)
=======================================================

rh-codebase, 2021-06-02:
In Python 3 for which this client is written against, string of characters default encoding is unicode. 2 bytes per character symbol. Python 2 on the other hand, a string of characters is an immutable byte array.  Bytes(actually bits) are sent on the bus. State that you want a Python 2 immutable byte array or python 2 string type.

This class isn't written for 1 person's use. The raw() method has value and will be kept. rawascii() will be added and will return a Python 2 immutable byte array known as a str type. If you want it mutable, then a Python 2 bytearray type is the only choice.

Conversion to upper case will be added as a different issue.

ldaddario, 2021-06-02:
I prefer not to respond further in this thread because I cannot tell who
is writing the other messages nor who is reading mine.

--Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
        +1/626/510-6492 (home)
        +1/626/657-6357 (mobile)
=======================================================

rh-codebase, 2021-06-04:
Fixed. Returns a Python 2 str type. Sched. for release v1.0.1

ldaddario, 2022-09-22:
This issue was closed long ago, but it was never fixed.  A small part of the issue was fixed by returning a byte string instead of a string of 2-char hex representions of those bytes.  But the main issue that the method fails to return all bytes of the response without interpretation has not been addressed.
1.  This method should always return without raising any exception.  If there is no response within the timeout period. the return value should be an empty string, not a timeout exception.  
2. The method should not attempt to interpret the first byte of the response as ACK or NAK; it should include that byte in the return string. In the case of NAK, it should include all bytes of the error code, without interpretation.
3. The return string should include the terminating CR byte.  If there is no CR received prior to the timeout, it should return all bytes received (without a CR).
4. In the outgoing (command) direction, all bytes that the caller provided should be sent without modification.  The only thing that the method should do is pre-pend the address byte and append the CR termination byte.  (I think this is what the method now does, but I'd like to be sure.)

closed 2023-03-29 by caseyjlaw