ovro-lwa/lwa-issues #89: openibdonboot service may not be working

rh-codebase, 2022-02-08:
After powering up GPU servers, marin needed to run:
 sudo /etc/init.d/openibd restart on gpus 01, 02, 03, 05, 07, 08.

On lxdlwagpu02:
`● openibdonboot.service - Load openibd on boot
   Loaded: loaded (/etc/systemd/system/openibdonboot.service; enabled; vendor preset: enabled)
   Active: active (exited) since Mon 2022-02-07 23:27:48 UTC; 24h ago
 Main PID: 2582 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/openibdonboot.service

Feb 07 23:27:47 lxdlwagpu02 systemd[1]: Starting Load openibd on boot...
Feb 07 23:27:47 lxdlwagpu02 root[2618]: openibd: running in manual mode
Feb 07 23:27:48 lxdlwagpu02 openibd[2582]: [49B blob data]
Feb 07 23:27:48 lxdlwagpu02 systemd[1]: Started Load openibd on boot.
Feb 07 23:28:11 lxdlwagpu02 root[4465]: openibd: Set node_desc for mlx5_1: lxdlwagpu02 HCA-2
(casa) ubuntu@lxdlwagpu02:~$ uptime
 23:40:09 up 1 day, 14 min,  1 user,  load average: 8.76, 8.35, 8.42
`
According to systemd, this ran w/out error.

Interesting that gpu04 didn't help.

marinanderson, 2022-03-30:
Following from a discussion with @rh-codebase during software meeting on 03/30 -- confirming that this does not work on any of the gpu nodes after a restart. Running sudo /etc/init.d/openibd restart is still necessary in order to get the gpu pipelines running.

Potential fixes (band-aids in Rick's words?) include running via a cronjob at restart, or packaging into the correlator start script. Are there preferences? I could do the former. Would need @realtimeradio for the latter.

rh-codebase, 2022-03-30:
I would be surprised if the cron fixes this unless it is run on a rather frequent cadence. Also, It seems like restarting(which reloads the drivers) fails if the pipeline is running and using the drivers. This makes sense but having failing cronjobs isn't elegant either so my preference is to do this in the pipeline start code when it is needed.

rh-codebase, 2022-03-30:
Another thought is perhaps the systemd script is being run at the wrong time during the boot and there's a slight race explaining why gpu04 was lucky. I'll look into modifying the boot script to run later in the boot process.

rh-codebase, 2022-03-30:
I made some changes to the openibdonboot service on lxdlwagpu01. It should run at the end of the boot process. Reboot lxdlwagpu01 and test the pipeline and update results here.

yupinghuang, 2022-03-31:
Did any log message get written?

rh-codebase, 2022-04-01:
none that I could find. systemd reports that openibd and openibdonboot ran w/out error. The former is a service provided via the openibd package and openibdonboot is our service written to try to see if restarting /etc/init.d/openibd fixes the problem.

yupinghuang, 2022-04-01:
I remember the days when I used upstart and scripts that echo to stdout/stderr just gets lost and it was extraordinarily difficult to figure out what was going on. It may be worth trying to pipe the output of /etc/init.d/openibd to a file in your systemd service and see if anything happens.

yupinghuang, 2023-03-03:
So after the power down/up a few days a go all GPU nodes (except for 02)'s pipelines refused to come up and running `sudo /etc/init.d/openibd restart` and then restarting the ectd service did help.

....so I guess this was never resolved?

rh-codebase, 2023-03-03:
yep. never resolved. need more info from systemd as to why this service doesn't work.

rh-codebase, 2023-03-08:
possible workaround: place sudo /etc/init.d/openibd restart in the startup code for the pipelines. The pipelines should know whether they have come up or not, and if not, run the above command and try again.