ovro-lwa/lwa-issues #90: Data recorder deployment

jaycedowell, 2022-02-09:
To get the data recorder software on lxdlwagpu09 today I did a:
 ```
cd ~
git clone https://github.com/lwa-project/ovro_data_recorder
cd ~/proj/lwa-shell/mnc_python/
python setup.py install
cd ~/proj/lwa-shell/caltech-bifrost-dsp
*fix bifrost*
make -j all
cd python
python setup.py install
sudo ldconfig -v
cd ~/ovro_data_recorder/services
./generate_services.py
sudo cp dr-vslow-*.service /etc/systemd/system/
sudo cp dr-beam-1.service /etc/systemd/system/
sudo systemctl daemon-reload
cd ..
./update_casa_data.sh
```
A couple of things to note about this:
 * `*fix bifrost*` and the related Bifrost rebuild/reinstall is something that I need to work with @realtimeradio on so that it is no longer a problem
 * the `update_casa_data.sh` is something that I currently do as part of the .service files but it should probably only be done once at system startup
 * in the future when the data recording is not done on lxdlwagpu09 the Bifrost build will probably need to be tweaked to turn off CUDA support if the machines do not have CUDA installed

yupinghuang, 2022-04-04:
I'm trying to a) enable `--no-tar` for the slow visibilities, and b) set this up on the calim test node.

Permission to regenerate the slow visibility service files with the `--no-tar` argument and then overwriting existing ones on gpu09?

jaycedowell, 2022-04-04:
I don't have any problem with that.  If you are trying to reproduce my setup the `*fix bifrost*`/make/install steps shouldn't be required if you have a recent checkout of the caltech-bifrost-dsp repo.

jaycedowell, 2022-08-04:
Replaced by #135.

closed 2022-08-04 by jaycedowell