ovro-lwa/lwa-issues #91: DsaStore attribute error in lwa-pyutils for ARX client

caseyjlaw, 2022-02-12:
I am on lxdlwacr and am trying to run some ARX monitoring in Python. I can import the arx module, but it fails to run a method to get configuration.

Seems like a version difference. I pulled the latest dsa-pyutils and lwa-pyutils and reinstalled. No change.

```
ma = lwa_arx.ARX()

/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/conf/etcdConfig.yml

ma.get_all_chan_cfg(32)

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-8-09f164fffed4> in <module>
----> 1 ma.get_all_chan_cfg(32)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in get_all_chan_cfg(self, arx_addr, user_timeout)
    926 
    927         """
--> 928         chan_cfgs = self._get_all_chan_cfg(arx_addr, user_timeout)
    929         #self._show_chan_cfg(0, chan_cfg)
    930         rtn = []

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _get_all_chan_cfg(self, arx_addr, user_timeout)
   1040         """
   1041 
-> 1042         rtn = self._send(arx_addr, 'geta', '', user_timeout)
   1043 
   1044         self._check_rtn(rtn, SET_CHAN_CFG_ERRORS)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _send(self, arx_addr, cmd, val, user_timeout)
    386 
    387         cmd_key = '{}{}'.format(CMD_KEY_BASE, arx_addr)
--> 388         rtn = self.my_cr.send(cmd_key, cmd, val, user_timeout)
    389 
    390         return rtn

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/cmd_rsp.py in send(self, key, cmd, val, timeout)
    176                 raise snre.ServiceNoResponseException()
    177         else:
--> 178             self.my_store.delete(self.full_response_key)
    179             return self.response_d

AttributeError: 'DsaStore' object has no attribute 'delete'

```

rh-codebase, 2022-02-12:
add output of:

`pip list | grep utils`

on cr machine.

caseyjlaw, 2022-02-12:
```
(casa) ubuntu@lxdlwacr:~$ pip list | grep utils
dsa110-pyutils          v3.4.2-dirty
ipython-genutils        0.2.0
lwa-pyutils             v1.4.1-dirty
python-utils            2.5.6
```

rh-codebase, 2022-02-12:
My lwa-pyutils has v1.4.2.

caseyjlaw, 2022-02-12:
I see the issue is with dsa-pyutils being out of date. Updating that to the latest tag fixes the issue.

closed 2022-02-12 by caseyjlaw