ovro-lwa/lwa-issues #94: bokeh dashboard unable to open websocket

rh-codebase, 2022-03-05:
output in the screen session 'dash' running on lxdlwacr shows while attaching with a browser after setting up the following tunnel within ovro
Tunneling to port 5006 on lxdlwacr via lwamaas

2022-03-05 00:50:37,099 Triggering stats polling loop stop
Trying to stop a non-existent command watch
2022-03-05 00:50:37,099 Trying to stop a non-existent command watch
2022-03-05 00:50:37,355 Got command response
2022-03-05 00:50:37,356 Response: {'id': '1646441437346861', 'val': {'status': 'normal', 'response': 23039, 'timestamp': 1646441437.3532531}}
2022-03-05 00:50:37,368 Got command response
2022-03-05 00:50:37,368 Response: {'id': '1646441437361429', 'val': {'status': 'normal', 'response': 23040, 'timestamp': 1646441437.3669345}}
2022-03-05 00:50:37,379 Got command response
2022-03-05 00:50:37,380 Response: {'id': '1646441437374166', 'val': {'status': 'normal', 'response': 23041, 'timestamp': 1646441437.3773813}}
2022-03-05 00:50:37,394 Got command response
2022-03-05 00:50:37,395 Response: {'id': '1646441437385031', 'val': {'status': 'normal', 'response': 23042, 'timestamp': 1646441437.3930023}}
2022-03-05 00:50:37,410 Got command response
2022-03-05 00:50:37,411 Response: {'id': '1646441437403229', 'val': {'status': 'normal', 'response': 17219, 'timestamp': 1646441437.4079015}}
2022-03-05 00:50:37,424 Got command response
2022-03-05 00:50:37,425 Response: {'id': '1646441437418290', 'val': {'status': 'normal', 'response': 17220, 'timestamp': 1646441437.4225235}}
2022-03-05 00:50:37,436 Got command response
2022-03-05 00:50:37,436 Response: {'id': '1646441437429873', 'val': {'status': 'normal', 'response': 17221, 'timestamp': 1646441437.434088}}
2022-03-05 00:50:37,462 Got command response
2022-03-05 00:50:37,463 Response: {'id': '1646441437441407', 'val': {'status': 'normal', 'response': 17222, 'timestamp': 1646441437.461497}}
2022-03-05 00:50:37,479 Got command response
2022-03-05 00:50:37,479 Response: {'id': '1646441437469504', 'val': {'status': 'normal', 'response': 21494, 'timestamp': 1646441437.4774284}}
2022-03-05 00:50:37,486 Got command response
2022-03-05 00:50:37,486 Response: {'id': '1646441437481933', 'val': {'status': 'normal', 'response': 21495, 'timestamp': 1646441437.4850276}}
2022-03-05 00:50:37,496 Got command response
2022-03-05 00:50:37,497 Response: {'id': '1646441437490351', 'val': {'status': 'normal', 'response': 21496, 'timestamp': 1646441437.4933853}}
2022-03-05 00:50:37,511 Got command response
2022-03-05 00:50:37,511 Response: {'id': '1646441437502291', 'val': {'status': 'normal', 'response': 21497, 'timestamp': 1646441437.5055737}}
2022-03-05 00:50:37,528 Got command response
2022-03-05 00:50:37,529 Response: {'id': '1646441437522402', 'val': {'status': 'normal', 'response': 30504, 'timestamp': 1646441437.5257862}}
2022-03-05 00:50:37,540 Got command response
2022-03-05 00:50:37,540 Response: {'id': '1646441437535638', 'val': {'status': 'normal', 'response': 30505, 'timestamp': 1646441437.5390475}}
2022-03-05 00:50:37,589 Got command response
2022-03-05 00:50:37,589 Response: {'id': '1646441437544503', 'val': {'status': 'normal', 'response': 30506, 'timestamp': 1646441437.5856555}}
2022-03-05 00:50:37,600 Got command response
2022-03-05 00:50:37,601 Response: {'id': '1646441437593879', 'val': {'status': 'normal', 'response': 30507, 'timestamp': 1646441437.5974844}}
2022-03-05 00:50:37,613 Got command response
2022-03-05 00:50:37,613 Response: {'id': '1646441437605707', 'val': {'status': 'normal', 'response': 23104, 'timestamp': 1646441437.6100266}}
2022-03-05 00:50:37,624 Got command response
2022-03-05 00:50:37,624 Response: {'id': '1646441437617838', 'val': {'status': 'normal', 'response': 23105, 'timestamp': 1646441437.621927}}
2022-03-05 00:50:37,635 Got command response
2022-03-05 00:50:37,635 Response: {'id': '1646441437628379', 'val': {'status': 'normal', 'response': 23106, 'timestamp': 1646441437.6329308}}
2022-03-05 00:50:37,646 Got command response
2022-03-05 00:50:37,647 Response: {'id': '1646441437639536', 'val': {'status': 'normal', 'response': 23107, 'timestamp': 1646441437.6436641}}
2022-03-05 00:50:37,657 Got command response
2022-03-05 00:50:37,657 Response: {'id': '1646441437651270', 'val': {'status': 'normal', 'response': 30434, 'timestamp': 1646441437.6548867}}
2022-03-05 00:50:37,668 Got command response
2022-03-05 00:50:37,668 Response: {'id': '1646441437662317', 'val': {'status': 'normal', 'response': 30435, 'timestamp': 1646441437.6665072}}
2022-03-05 00:50:37,678 Got command response
2022-03-05 00:50:37,678 Response: {'id': '1646441437672757', 'val': {'status': 'normal', 'response': 30436, 'timestamp': 1646441437.6760771}}
2022-03-05 00:50:37,694 Got command response
2022-03-05 00:50:37,695 Response: {'id': '1646441437684194', 'val': {'status': 'normal', 'response': 30437, 'timestamp': 1646441437.6929111}}
2022-03-05 00:50:37,706 Got command response
2022-03-05 00:50:37,706 Response: {'id': '1646441437699882', 'val': {'status': 'normal', 'response': 15509, 'timestamp': 1646441437.7040102}}
2022-03-05 00:50:37,718 Got command response
2022-03-05 00:50:37,719 Response: {'id': '1646441437712578', 'val': {'status': 'normal', 'response': 15510, 'timestamp': 1646441437.7173324}}
2022-03-05 00:50:37,729 Got command response
2022-03-05 00:50:37,729 Response: {'id': '1646441437723397', 'val': {'status': 'normal', 'response': 15511, 'timestamp': 1646441437.7267668}}
2022-03-05 00:50:37,742 Got command response
2022-03-05 00:50:37,742 Response: {'id': '1646441437733844', 'val': {'status': 'normal', 'response': 15512, 'timestamp': 1646441437.7397804}}
2022-03-05 00:50:38,277 Refusing websocket connection from Origin 'http://localhost:25006';                       use --allow-websocket-origin=localhost:25006 or set BOKEH_ALLOW_WS_ORIGIN=localhost:25006 to permit this; currently we allow origins {'localhost:5006'}
2022-03-05 00:50:38,277 403 GET /LWA_dashboard/ws (10.40.0.4) 1.04ms

Looks like bokeh needs to know the destination port which is out of its control. Hmm. That seems odd.

caseyjlaw, 2022-03-05:
I can reproduce this. I added the environment variable when running the server and no longer see that error. However, I got a warning message about a different port (the one used for http traffic).
Can you confirm that it works for you?

rh-codebase, 2022-03-05:
I didn't see your prev. message in time so I tried setting up the my local port to 5006 and it now works. going back to try a different local port now. Does the local port need to be 25006 or can it now be anything?

rh-codebase, 2022-03-05:
all looks good here. I setup a tunnel to local port 15006 and worked. thx. closing.

closed 2022-03-05 by rh-codebase