ovro-lwa/lwa-issues #96: ArxException: Unable to parse json error msg for an empty arx slot

yupinghuang, 2022-04-01:
Marin says that 44 is out for repairs so the slot is empty.
```
In [6]: for adr in adrs:
   ...:     print(adr)
   ...:     ma.load_cfg(adr, 2)
   ...:     
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
33
34
35
36
37
38
39
40
41
42
43
44
---------------------------------------------------------------------------
JSONDecodeError                           Traceback (most recent call last)
~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _check_rtn(self, rtn, errors)
    305                 # print("_check_rtn: err_msg_json: {}".format(err_msg_json))
--> 306                 err = json.loads(err_msg_json)
    307             except:

~/anaconda3/envs/casa/lib/python3.6/json/__init__.py in loads(s, encoding, cls, object_hook, parse_float, parse_int, parse_constant, object_pairs_hook, **kw)
    353             parse_constant is None and object_pairs_hook is None and not kw):
--> 354         return _default_decoder.decode(s)
    355     if cls is None:

~/anaconda3/envs/casa/lib/python3.6/json/decoder.py in decode(self, s, _w)
    338         """
--> 339         obj, end = self.raw_decode(s, idx=_w(s, 0).end())
    340         end = _w(s, end).end()

~/anaconda3/envs/casa/lib/python3.6/json/decoder.py in raw_decode(self, s, idx)
    354         try:
--> 355             obj, end = self.scan_once(s, idx)
    356         except StopIteration as err:

JSONDecodeError: Expecting ',' delimiter: line 1 column 27 (char 26)

During handling of the above exception, another exception occurred:

ArxException                              Traceback (most recent call last)
<ipython-input-6-357f27a67f4f> in <module>()
      1 for adr in adrs:
      2     print(adr)
----> 3     ma.load_cfg(adr, 2)
      4 

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in load_cfg(self, arx_addr, loc, user_timeout)
   1564         rtn = self._send(arx_addr, 'load', loc_str, user_timeout)
   1565 
-> 1566         return self._check_rtn(rtn, LOAD_ERRORS)
   1567 
   1568     def save_cfg(self,

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/lwa_arx.py in _check_rtn(self, rtn, errors)
    306                 err = json.loads(err_msg_json)
    307             except:
--> 308                 raise ARXE.ArxException("Unable to parse json error msg")
    309             if errors is not None and err['ERR'] == 'NAK':
    310                 err_msg = "{} {} - {}".format(err['ERR'], err['MSG'],

ArxException: Unable to parse json error msg
```

rh-codebase, 2022-04-01:
how is the code populating 'adrs'? It should be getting valid addresses from the python API which reads the current system config from etcd. If the system config is out of date, then this is a bug. Otherwise, lwa_arx is performing as designed by throwing an exception.

yupinghuang, 2022-04-01:
Shouldn't it throw an exception that's a bit more informative? This is obviously low priority but it seems like the json parsing error occurred while it tried to throw the actual exception.

rh-codebase, 2022-04-01:
Our comments crossed. If you find the system config needs updating, let's file that separately.

rh-codebase, 2022-04-01:
Oh. I forgot to ask what version of the lwa-pyutils are you using:  pip list | grep pyutils  should show it.

yupinghuang, 2022-04-01:
```
(casa) ubuntu@lxdlwagpu09:~$ pip list | grep pyutils
dsa110-pyutils               v3.5.3-10-g4df687a-dirty
lwa-pyutils                  v1.4.2-dirty
```
I didn't get the adrs from the API so I will do that and see if it's up-to-date

rh-codebase, 2022-04-19:
Fixed in version 1.3.4 of the ARX service. Example code:

closed 2022-04-19 by rh-codebase