ovro-lwa/lwa-issues #97: Issues with lwa-shell on lxdlwagpu04, 06, 08, and 09.

marinanderson, 2022-04-01:
Something funny happened with lwa-shell on restart on the above gpu nodes, and some directories/files are incomplete/missing.

Yuping found the following:
(casa) ubuntu@lxdlwagpu09:~/proj/lwa-shell/mnc_python$ git status
On branch cjl/dev
Your branch is up to date with 'origin/cjl/dev'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	deleted:    .gitignore
	deleted:    LICENSE
	deleted:    README.md
	deleted:    config/lwa_corr_config.yaml
	deleted:    mnc/__init__.py
	deleted:    mnc/common.py
	deleted:    mnc/control.py
	deleted:    mnc/ezdr.py
	deleted:    mnc/lwa_hiplot.py
	deleted:    mnc/mcs.py
	deleted:    mnc/xengine_beamformer_control.py
	deleted:    notebooks/LWA_MnC_prototyping.ipynb
	deleted:    notebooks/LWA_autocorrelations_to_uvdata.ipynb
	deleted:    notebooks/LWA_dashboard.ipynb
	deleted:    notebooks/lwa_hiplot.ipynb
	deleted:    scripts/FXstart.py
	deleted:    scripts/astm_monitor.py
	deleted:    setup.py
	deleted:    version.py

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	build/
	dist/
	mnc_python.egg-info/
	notebooks/.ipynb_checkpoints/

Looks like files were deleted for some reason? Anyway, as a result, pipelines can't run on 4, 6 and 8 and data recorder is impacted on 09.

yupinghuang, 2022-04-01:
Actually, all of the repos show the same things. Top-level directories exist but all the files are deleted.

caseyjlaw, 2022-04-01:
I think the caltech-lwa repo on lwacr is also missing files. Related?

rh-codebase, 2022-04-01:
This is an easy fix(git checkout main). I'll fix on those hosts. For some reason , those hosts had the wrong remote when I deployed new software updates for lwa-pyutils so this is related to that change. I'll close issue when those hosts have been fixed.

yupinghuang, 2022-04-01:
I can just do git checkout -- since I'm working on those hosts. Are they gonna have the right remotes going forward?

yupinghuang, 2022-04-01:
Did
```
for d in ~/proj/lwa-shell/*/;do cd $d && git checkout -- *; done
```
on gpu04, 06, 08, 09, lxdlwacr. Should be good now?

rh-codebase, 2022-04-01:
yikes. we need to coordinated when fixing up these dirs. The reason is related to myrepos and how .mrconfig_production was defined. That has been fixed and pushed as v4.3.2 in lwa-shell. running mr using the .mrconfig_production file now works as expected. A lot of our repos defined in this file do not have version tags. I'll create an issue for getting around to versioning our repos.  Next time though, the assignee of the issue should be the one doing the changes. Feel free to reassign though.  Regarding caltech-bifrost-dsp. I think Jack made an update yesterday so rebuilding/installing  may be necessary. It is checked out at the correct commit(572af54)

closed 2022-04-01 by rh-codebase