ovro-lwa/lwa-issues #102: powermon errors on SNAP6

marinanderson, 2022-04-06:
The following power monitoring error previously appeared on snap06 and snap09:

`ERROR - powermon - Failed to instantiate I2C control object`

Mark re-seated the SD cards on both boards, then we rebooted the boards via power cycling of the SNAP chassis.

Fixed things for snap09, but still getting that same error on snap06:
`2022-04-06 13:55:19,827 - lwa_f.blocks.block:snap06 - ERROR - powermon - Failed to instantiate I2C control object`

rh-codebase, 2022-04-06:
Perhaps try swapping the SD cards between the 2 SNAPS and see if the issue follows the card.

Princemongo, 2022-04-06:
Mark here - were both chassis power cycled? SNAP06 is in chassis #1 while SNAP09 is in chassis #2.

marinanderson, 2022-04-06:
Yup, we power cycled both chassis.

Princemongo, 2022-04-06:
@jack-h Can the microSD cards be removed and inserted with the power on? Also, can the microSD cards be swapped between SNAPs?

jack-h, 2022-04-07:
> @jack-h <https://github.com/jack-h> Can the microSD cards be removed and
> inserted with the power on?
They can be inserted and removed, but the firmware they contained is only
copied at boot, so a power cycle is required after inserting a new card.

Also, can the microSD cards be swapped between SNAPs?

Yes!

Princemongo, 2022-04-12:
Has snap06 magically fixed itself or is there still a problem?

Princemongo, 2022-04-15:
I took the microSD cards from snap06 and snap05 and when I grabbed the microSD-to-USB adapter, it had an identical card in it! I copied the bin file from snap05 by putting it on my Windows box and then overwriting the files on the other two microSD cards. When I inserted them back into the snaps, now neither works! I'm thinking it might be a mistake to use Windows. I will repeat the process tomorrow but using the microSD card from snap04 as the "master".

marinanderson, 2022-04-16:
No longer getting the powermon error on snap06, so Mark's trick with the microSD card from snap04 must have worked. 🎉

closed 2022-04-16 by marinanderson