ovro-lwa/lwa-issues #103: Non-zero missing data fraction for beamformer

marinanderson, 2022-04-06:
See output from the dr-beam-1 log below. Missing data fraction is consistently 3.125%.

> (casa) ubuntu@lxdlwagpu09:~/proj/lwa-shell/mnc_python/mnc$ tail -f ~/log/dr-beam-1.lxdlwagpu09.log
>2022-04-06 22:55:59 [DEBUG   ]  active operation: False
>2022-04-06 22:55:59 [DEBUG   ] ===   ===
>2022-04-06 22:56:08 [INFO    ] HDF5Record - Scheduled recording for 2022-04-06 22:56:23.366852 to 2022-04-06 >22:56:53.366852 to /home/ubuntu/data/beam01/059675_225623366852bf11a30
>2022-04-06 22:56:09 [DEBUG   ] === Performance Report ===
>2022-04-06 22:56:09 [DEBUG   ]  max acquire/process/reserve times: 1.001/1.001/0.000
>2022-04-06 22:56:09 [DEBUG   ]  receive data rate: 47011971.337 B/s
>2022-04-06 22:56:09 [DEBUG   ]  missing data fraction: 3.125%
>2022-04-06 22:56:09 [DEBUG   ]  pipeline lag: 2.135294
>2022-04-06 22:56:09 [DEBUG   ]  load average: 0.12, 0.12, 0.27
>2022-04-06 22:56:09 [DEBUG   ] ===   ===

jack-h, 2022-04-12:
Presumably exactly one X-engine pipeline (of the 32). Does "consistently" here mean on multiple X-eng intialization cycles?

Is the series of commands to reproduce this scriptified and/or is just the standard beamformer recipe?

jaycedowell, 2022-04-12:
[ovro_data_recorder/f125e3a](https://github.com/lwa-project/ovro_data_recorder/commit/f125e3a342bc8d20167238b5c7ef1f891748b0f9) adds server and pipeline/subband boundaries to the spectra plots.

jaycedowell, 2022-04-12:
[ovro_data_recorder/08ee974](https://github.com/lwa-project/ovro_data_recorder/commit/08ee97405965bd8c46cd5b101d2f01838cc946bc) adds a utility to make it easy to pull out the latest power beam spectra.

jaycedowell, 2022-04-12:
Those two commits should make it easier to tell what is going on.  A missing subband (or all-zero data) will show up as a break in the spectra between two of the vertical dashed/dotted lines.

iveydavis, 2022-05-05:
Marin and I ran the beams for 60s intervals. Tried 3 different combos (beams 1 and 2, beams 1, 2, and 3, and all four beams) to see if # of beams used affected anything. Beam 2 was consistently missing 6.25% and beam 4 was missing 3.125%.

iveydavis, 2022-05-12:
Sorry for the late reply Jack-- was double checking the logs to make sure i was looking at the right images since i apparently misread them previously to find the data losses. Here's the XX dynamic spectra for the different sets of beams
[beam1_2_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682873/beam1_2_concurrent_beams.pdf)
[beam1_3_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682874/beam1_3_concurrent_beams.pdf)
[beam1_4_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682875/beam1_4_concurrent_beams.pdf)
[beam2_2_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682890/beam2_2_concurrent_beams.pdf)
[beam2_3_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682891/beam2_3_concurrent_beams.pdf)
[beam2_4_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682892/beam2_4_concurrent_beams.pdf)
[beam3_4_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682901/beam3_4_concurrent_beams.pdf)
[beam3_3_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682902/beam3_3_concurrent_beams.pdf)
[beam4_4_concurrent_beams.pdf](https://github.com/ovro-lwa/lwa-issues/files/8682904/beam4_4_concurrent_beams.pdf)

jack-h, 2022-05-15:
Thanks. Digging deeper, it looks like the problem is that the missing data is never sent. Bifrost claims it is, but the network switch is adament it isn't.

I have made some minor tweaks to the beamformer transmission code, and 3 separate attemps to send a single beam  (stopping and starting the gpu pipelines between attempts) have resulted in 0% packet loss.
But, sometimes I could get no loss even with the original code, so I'm not sure if this is fixed or I was just temporarily lucky.

jaycedowell, 2022-08-04:
Since this seems to be resolved I'm going to go ahead and close this out.

yupinghuang, 2023-02-24:
Saw 31.25% data loss on dr1 when I ran dr1 and dr2 together on zenith. Gonna leave it to y'all the pros

jack-h, 2023-02-24:
Can you send me the series of steps you took to hit this issue (specifically, was the correlator / beamformer already running, or did you initialize everything from scratch, or did you init multiple times, etc.).
Looking back at this issue clearly I could never recreate the problem on demand :-S

yupinghuang, 2023-02-24:
Here's all I did (from ipython; first number is session and second is the order of command).

I did not do anything with the F engine. And `configure_xengine` does stop and then start the x engine. So it went like
- restarted x engine with beam1 and beam2 packets being sent -- beam 1 claimed ~50% packet loss
- restarted the DR services
- restarted x engine with beam 1 and beam 2 packets being sent -- beam 1 claimed ~37.5% packet loss

```
652/1: from mnc.control import Controller
652/2: con = Controller(f"/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml")
652/3: con.configure_xengine(['drvs', 'dr1', 'dr2'])
652/4: con.control_bf(num=1, targetname='zenith',track=False)
652/5: con.control_bf(num=2, targetname='zenith',track=False)
652/6: con.status_dr()
652/7: con.status_dr()
652/8: con.start_dr(['drvs', 'dr1', 'dr2'])
652/9: con.start_dr(['drvs', 'dr1', 'dr2'], duration=4e7, time_avg=2000)
652/10: con.status_dr()
652/11: con.status_dr()
652/12: con.status_dr()
653/1: from mnc.control import Controller
653/2: con = Controller(f"/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml")
653/3: con.status_dr()
653/4: con.status_xengine()
653/5: con.status_dr()
653/6: con.status_dr()
653/7: con.configure_xengine(['drvs', 'dr1', 'dr2'])
653/8: con.status_dr()
653/9: con.status_dr()
653/10: con.status_dr()
653/11: con.status_dr()
653/12: con.status_dr()
653/13: con.start_dr(['dr1', 'dr2'], duration=4e7, time_avg=2000)
653/14: con.status_dr()
653/15: con.status_dr()
653/16: con.status_dr()
653/17: con.status_dr()
653/18: con.status_dr()
653/19: con.status_dr()
655/1: from mnc.control import Controller
655/2: con = Controller(f"/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml")
655/3: con.start_dr(['dr1', 'dr2'], duration=4e7)
655/4: con.stop_dr(['dr1', 'dr2'], duration=4e7)
655/5: con.stop_dr(['dr1', 'dr2'])
655/6: con.status_dr()
655/7: con.status_dr()
655/8: con.start_dr(['dr1', 'dr2'], duration=4e7)
655/9: con.status_dr()
655/10: con.status_dr()
655/11: con.status_dr()
655/12: con.status_dr()
```

Note that I haven't done anything since then apart from stopping the recording operation. data recorders are still listening to the ports (I think). beam 1 does report a proportionately smaller throughput than beam 2 but the wacky thing is that `like_bmon.py` on calim01 doesn't show missing bytes (or late bytes) increasing but `dr-beam-1` seemed to think that it wasn't getting good packets.
```
(datarecorder) [pipeline@lwacalim01 log]$ tail -f dr-beam-1.lwacalim01.log
2023-02-24 16:56:50 [DEBUG   ]  missing data fraction: 31.250%
2023-02-24 16:56:50 [DEBUG   ]  pipeline lag: 1.96623
2023-02-24 16:56:50 [DEBUG   ]  load average: 0.33, 0.31, 0.39
2023-02-24 16:56:50 [DEBUG   ] ===   ===
2023-02-24 16:56:50 [DEBUG   ] === Status Report ===
2023-02-24 16:56:50 [DEBUG   ]  summary: error
2023-02-24 16:56:50 [DEBUG   ]  info: Packet loss during receive >10% (31.2% missing)
2023-02-24 16:56:50 [DEBUG   ]  queue size: 0
2023-02-24 16:56:50 [DEBUG   ]  active operation: False
2023-02-24 16:56:50 [DEBUG   ] ===   ===
2023-02-24 16:57:00 [DEBUG   ] === Performance Report ===
2023-02-24 16:57:00 [DEBUG   ]  max acquire/process/reserve times: 1.007/1.004/0.001
2023-02-24 16:57:00 [DEBUG   ]  receive data rate: 33435745.394 B/s
2023-02-24 16:57:00 [DEBUG   ]  missing data fraction: 31.250%
2023-02-24 16:57:00 [DEBUG   ]  pipeline lag: 1.96623
2023-02-24 16:57:00 [DEBUG   ]  load average: 0.43, 0.33, 0.40
2023-02-24 16:57:00 [DEBUG   ] ===   ===
^C
(datarecorder) [pipeline@lwacalim01 log]$ tail -f dr-beam-2.lwacalim01.log
2023-02-24 16:57:00 [DEBUG   ]  queue size: 0
2023-02-24 16:57:00 [DEBUG   ]  active operation: False
2023-02-24 16:57:00 [DEBUG   ] ===   ===
2023-02-24 16:57:11 [DEBUG   ] === Performance Report ===
2023-02-24 16:57:11 [DEBUG   ]  max acquire/process/reserve times: 0.998/0.998/0.000
2023-02-24 16:57:11 [DEBUG   ]  receive data rate: 48489893.865 B/s
2023-02-24 16:57:11 [DEBUG   ]  missing data fraction: 0.000%
2023-02-24 16:57:11 [DEBUG   ]  pipeline lag: 2.626796
2023-02-24 16:57:11 [DEBUG   ]  load average: 0.44, 0.34, 0.40
2023-02-24 16:57:11 [DEBUG   ] ===   ===
```

yupinghuang, 2023-02-24:
I think this might be because of me trying to do averaging and something having to do with https://github.com/ovro-lwa/lwa-issues/issues/264 though.....

yupinghuang, 2023-02-24:
Er no. The latest start on dr1 didn't have any averaging set.

jack-h, 2023-02-24:
Thanks for the thorough recipe!

yupinghuang, 2023-02-24:
[image]
[image]
@nkosogor observed and @iveydavis plotted the data and it didn't seem like there was packet loss even though the data recorder said so.

jack-h, 2023-02-24:
The plot thickens

akvydula, 2023-02-28:
I was trying some simple slow visibility tests yesterday following the system use recipe and saw some warnings that might be relevant to this discussion:

> con.status_dr(['dr1', 'drvs'])  
> Out[21]: 
> ['idle',
>  'WARNING: dr1 not fully operational: Only 8 of 9 threads active - MainThread,etcd3_watch_7f41c0ee0a90,Thread-1,Thread-2,CaptureOp,SpectraOp,StatisticsOp,GlobalLogger',
>  'recording']

akvydula, 2023-02-28:
@yupinghuang Yes, 128 ms as described in the system use recipe.

yupinghuang, 2023-02-28:
ok! That's a separate issue https://github.com/ovro-lwa/lwa-issues/issues/264 once Jayce's change is deployed that should be good.

caseyjlaw, 2023-03-29:
Given Jack's successful test and recent success in beamforming observing, I'll suggest we close this.

closed 2023-03-29 by caseyjlaw