ovro-lwa/lwa-issues #99: Error encountered in xengine_beamformer_control during track

dgary50, 2022-04-06:
Marin suggested I note this error here:
When trying to track the Sun (e.g.)
    c = BeamPointingControl(1)
    t = BeamTracker(c, update_interval=15)
    t.track('Sun', duration=30)
we got this error (the last of a long line, but I guess this is the important one):
~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/blocks/beamform_control.py in update_delays(self, beam_id, delays, amps, load_time, time_unit)
     84                 load_adc_sample = int(load_time * self.fs_hz)
---> 85                 load_sample = load_adc_sample // (2*nchan)
     86             else:

NameError: name 'nchan' is not defined

Then it says in handling this error it gave another error:
~/ovro_data_recorder/xengine_beamformer_control.py in __exit__(self, exc_type, exc_value, exc_traceback)
     47     def __exit__(self, exc_type, exc_value, exc_traceback):
     48         if exc_type:
---> 49             warnings.warn(f"Failed to command {self.pipeline.host} (pipeline {self.pipeline.pipeline_id})")
     50         return True
     51
which suggests that the pipeline object is not correct.

Marin thinks this is a question for Jack.

marinanderson, 2022-04-06:
Still getting this error when running t.track('sun'). Perhaps we should have done some tracking during the beamformer test this morning.

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/blocks/beamform_control.py in update_delays(self, beam_id, delays, amps, load_time, time_unit)
     84                 load_adc_sample = int(load_time * self.fs_hz)
---> 85                 load_sample = load_adc_sample // (2*nchan)
     86             else:

NameError: name 'nchan' is not defined

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
<ipython-input-8-3ef813a55157> in <module>()
----> 1 t.track('sun')

~/ovro_data_recorder/xengine_beamformer_control.py in track(self, target_or_ra, dec, duration, start_time, verbose)
    475 
    476                 ## Point
--> 477                 self.control_instance.set_beam_pointing(az, alt, degrees=True, load_time=t_mark+2)
    478 
    479                 ## Find how much time we used and when we should sleep until

~/ovro_data_recorder/xengine_beamformer_control.py in set_beam_pointing(self, az, alt, degrees, load_time)
    345         # Apply
    346         for pol in range(NPOL):
--> 347             self.set_beam_delays(delays, pol, load_time=load_time)
    348 
    349     def set_beam_target(self, target_or_ra, dec=None, load_time=None, verbose=True):

~/ovro_data_recorder/xengine_beamformer_control.py in set_beam_delays(self, delays, pol, load_time)
    306         # Set the delays and amplitudes
    307         ptp = PipelineTaskPool(self.pipelines)
--> 308         ptp.beamform.update_delays(2*(self.beam-1)+pol, delays, amps, load_time=load_time, time_unit='time')
    309 
    310     def set_beam_pointing(self, az, alt, degrees=True, load_time=None):

~/ovro_data_recorder/xengine_beamformer_control.py in __call__(self, *args, **kwds)
     75             i = results[result]
     76             with AllowedPipelineFailure(self[i]):
---> 77                 output[i] = result.result()
     78         return output
     79 

~/ovro_data_recorder/xengine_beamformer_control.py in __exit__(self, exc_type, exc_value, exc_traceback)
     47     def __exit__(self, exc_type, exc_value, exc_traceback):
     48         if exc_type:
---> 49             warnings.warn(f"Failed to command {self.pipeline.host} (pipeline {self.pipeline.pipeline_id})")
     50         return True
     51 

AttributeError: 'function' object has no attribute 'host'

jaycedowell, 2022-04-12:
My portion of this (`AttributeError: 'function' object has no attribute 'host'`) should be fixed with [ovro_data_recorder/a649853](https://github.com/lwa-project/ovro_data_recorder/commit/a649853922805f28517f6800d04f5324823b1312).

jaycedowell, 2022-04-12:
I also updated the install of `lwa352_pipeline_control` on lxdlwagpu09 to pull in the fix for the `NameError: name 'nchan' is not defined` (thanks @jack-h).  I think that is everything.

jack-h, 2022-04-13:
Ah, sorry, I thought I already pulled that in on gpu09. So, are we calling this good to close?

jaycedowell, 2022-04-13:
Yes, I think this particular problem can be closed now.

closed 2022-04-13 by caseyjlaw

dgary50, 2022-04-13:
We are doing another solar beam test tomorrow and will confirm how it goes (or open a new report!). 

Dale

jack-h, 2022-10-11:
Are you running from the CR node or GPU9 (or somewhere else)?