ovro-lwa/lwa-issues #104: Create interface to show system config at any  time in system

rh-codebase, 2022-04-12:
System configuration is stored in git as a YAML file in the lwa-cfg repo each time a new config is pushed into etcd. This feature provides a way one can query the git repo and display the system config in some human readable form for any date.

caseyjlaw, 2022-04-12:
I was imagining something more like the "statusmon" influx stream at DSA-110. Alternatively, I could create a "time machine", also as done for DSA-110 via a python client to influx.
To do this, we'd need to be sure that the etcd key and payload are configured in a way that supports the science user.

ldaddario, 2022-04-14:
"This feature provides a way one can query the git repo and display the system config in some human readable form for any date."  Not really.  There are three steps:
- Do something physical to change the configuration.
- Put information about the change into the configuration spreadsheet.
- Push the spreadsheet into etcd. 
The repo knows only the time of the 3rd step, but it's the time of the 1st step that matters.

rh-codebase, 2022-04-14:
There are a few 'times' that people may want to track. Currently the system is only tracking the time when the System Config spreadsheet is updated(i.e. pushed into etcd) which changes the behavior of the software used to collect science data. The science data collection should be stopped prior to any physical changes and restarted after the new config is pushed into etcd. This is a workflow requirement. If the time of the first step mentioned previously is required, then please submit an enhancement issue to add a place in the System Configuration spreadsheet to record the time. Specify whether the time should represent the start of the change or when the change was complete or perhaps both are needed. 
A new spreadsheet has just been introduced which also records changes to the system known as 'The Log'. This issue was filed before this new log came into existence so for clarity, this issue is for the 'System Configuration Spreadsheet' only. There are currently no plans to parse 'The Log' into the system.

rh-codebase, 2022-04-14:
@caseyjlaw . The lwa-cfg repo contains the actual Excel spreadsheet as well as a YAML representation.  See the workflow description in the lwa-cfg repo. The Excel may be all that is needed. 'Human readable' is a bit ill defined but the Excel format does provide the same look as when the data were originally entered.

caseyjlaw, 2022-04-14:
I'm failing to recall the motivation for this feature. Can you post some example /cfg values and how someone would try to use a tool to query their history?

Incidentally, DSA-110 could use a feature like this, too. A scheme we develop here could be adapted there.

(Regarding Larry's point: I agree that it is reasonable to assume that most /cfg changes are done during down time, so the time offset is not critical.)

rh-codebase, 2022-04-14:
Dave or James brought this up during one of the meetings. In its simplest form, present the user with a human readable view of what the System Configuration was for a specified time.  Since the System Configuration spreadsheet is in Git, it's a matter of finding the closest ancestor to the specified time. 
Presenting everything under /cfg is much broader but could be combined with this.