ovro-lwa/lwa-issues #107: ADC initialization error on snap03

marinanderson, 2022-04-16:
Getting a lengthy error on SNAP03, I think these are the relevant parts:

2022-04-15 17:13:35,798 - lwa_f.blocks.block:snap03 - INFO - adc - FMC 1 board 1: Setting clock source to 1
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa_f/snap2_fengine.py in cold_start(self, program, initialize, test_vectors, sync, sw_sync, enable_eth, chans_per_packet, first_stand_index, nstand, macs, source_ip, source_port, dests)
    578             try:
--> 579                 self.adc.initialize(read_only=False)
    580             except RuntimeError:

...

~/anaconda3/envs/casa/lib/python3.6/site-packages/casperfpga-0.4.4.dev1077+head.e76da15-py3.6-linux-x86_64.egg/casperfpga/transport_tapcp.py in blindwrite(self, device_name, data, offset, use_bulk)
    609                 pass
--> 610         raise RuntimeError("Failed to write to register %s at offset %d" % (device_name, offset))
    611 

RuntimeError: Failed to write to register ads5296_spi_controller0 at offset 1

Have seen these errors before (on more than just SNAP03); @jack-h I think we talked about this at a software meeting a while back. Usually re-running `cold_start_from_config` one or more times works.  This time, I am unable to get SNAP03 to initialize.

jack-h, 2022-04-19:
`snap03` currently appears initialized OK (it claims to be happily sending
data). Did it magically come back to life?

jack-h, 2022-04-21:
After testing yesterday, my working theory is that this issue was due to
random communications collisions with the RFI monitoring routines. Once
these were stopped, things seemed to work fine. Will leave this issue open
until more data can be gathered.

The right long-term solution is probably for both the RFI script, and
general users, to use the etcd interface to the SNAPs

closed 2023-03-29 by caseyjlaw