ovro-lwa/lwa-issues #108: Better error when using data recorder monitor points

caseyjlaw, 2022-04-18:
Power beams can be discovered by ezdr.py:
```
> recorder = 'power'
> lwa_drc = ezdr.Lwa352RecorderControl(recorder)
Found 4 power recorders: dr1 dr2 dr3 dr4
```

But some data recorders have not been used yet. They have errors when accessing monitor points:
```
> drid = 'dr3'
> lwa_dr = mcs.Client(drid)
> rx_rate = lwa_dr.read_monitor_point('bifrost/rx_rate', id=drid)
ERROR2: the JSON object must be str, bytes or bytearray, not ‘NoneType’
```

It would be helpful to have more informative error in this case.

jaycedowell, 2022-04-19:
`mcs.Client` is currently setup so that it internally catches all exceptions that happen and prints out something like this "ERROR2:  ..." instead of letting the user get the relevant exception.  I originally did this to simplify the interface for users so that `lwa_dr.read_monitor_point('never_going_to_work', id=drid` would return `None` if the monitor point didn't exist rather than bother the user.  I thinking about having better error messages would it be preferred change this behavior and let the user deal with whatever exceptions are thrown?

caseyjlaw, 2022-04-19:
Handling exceptions is easy in Python and considered good form.
I would also be fine with returning `None` if no monitor point exists.

rh-codebase, 2022-04-19:
If no monitor points is considered an error state, then an Exception is the proper method to convey that. Otherwise, the application code needs to guard against None and constructs like:  for k,v in <monitor point dictionary>.items(): will break. If no monitor points is not an error, it would be best to return an empty dictionary in which case the for loop would just be skipped w/out any guards. An empty dictionary also satisfies the API in a strict sense.

jaycedowell, 2022-04-19:
> Handling exceptions is easy in Python and considered good form. I would also be fine with returning `None` if no monitor point exists.

The latter is the current behavior which would make the easy thing for me to comment out the `print("ERROR2: ...` and be done with it.  My question was more along the lines of what @rh-codebase brought up with regards to an API.  As far as I know there isn't a document that describes how M&C should interact with etcd and the subsystems so I don't know if a missing monitor point should be an error condition or not.  If this document exists I'd be happy to read it so I can implement something consistent with it.

rh-codebase, 2022-04-19:
My hope is that M&C python code interacts with etcd by using either the DsaStore class written for DSA in dsa110-pyutils repo or the CmdRsp class in lwa-pyutils repo. The public API returns a dictionary or throws if it cannot decode the etcd value into a dictionary or no response in the case of CmdRsp. An empty dictionary here is part of the happy path so no exceptions are thrown since an empty dictionary is a dictionary. The application code, can now decide if further Exceptions are to be thrown based on the requirements.  If the above two classes don't meet our needs then we should either add more or modify the existing ones. The real intent is to encapsulate etcd from all the application code so that it could be switched out for the next best thing in the future.

The only real requirement in regards to etcd is that values for the etcd key are in legal JSON format. The DsaStore class ensures this(or it's a bug). As such, they should always decode into a dictionary. Error conditions on the service side can be propagated back inside the JSON packet so clients are not entirely blind as to why something may have failed. The application libraries or applications themselves would be using the above classes and throwing their own exceptions. The CmdRsp class is an example as it uses the DsaStore class to communicate with etcd and just adds functionality to it.

Apologies for the winded response. I'll check that this is captured in one of the M&C docs. I believe I touched on this in some previous presentation slides but it does belong in an M&C doc. Along with this is a document describing additional requirements for monitor data as it relates to Influx and Grafana. Ingestion into Influx must be efficient as this aggregates the monitor data from all subsystems and must keep up. It's not the place to be rearranging data.

jaycedowell, 2022-04-19:
I didn't know about either DsaStore or CmdRsp.  That's why I ended up writing mcs.py with its MonitorPoint and Client classes.

closed 2023-02-22 by yupinghuang