ovro-lwa/lwa-issues #110: etcd connection failure

marinanderson, 2022-04-25:
Cannot connect to etcd from any of the gpu nodes? See full error message below:

In [1]: from lwa352_pipeline_control import Lwa352CorrelatorControl
   ...: p = Lwa352CorrelatorControl( ['lxdlwagpu01'], 
   ...: npipeline_per_host=4)
   ...: 
   ...: 
---------------------------------------------------------------------------
_MultiThreadedRendezvous                  Traceback (most recent call last)
~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py in handler(*args, **kwargs)
     45             try:
---> 46                 return f(*args, **kwargs)
     47             except grpc.RpcError as exc:

~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py in add_watch_callback(self, *args, **kwargs)
    569         try:
--> 570             return self.watcher.add_callback(*args, **kwargs)
    571         except queue.Empty:

~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py in add_callback(self, key, callback, range_end, start_revision, progress_notify, filters, prev_kv)
    104                 if new_watch.err:
--> 105                     raise new_watch.err
    106             finally:

~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/watch.py in _run(self)
    127             try:
--> 128                 for rs in response_iter:
    129                     self._handle_response(rs)

~/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py in __next__(self)
    425     def __next__(self):
--> 426         return self._next()
    427 

~/anaconda3/envs/casa/lib/python3.6/site-packages/grpc/_channel.py in _next(self)
    825                 elif self._state.code is not None:
--> 826                     raise self
    827 

_MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "failed to connect to all addresses"
	debug_error_string = "{"created":"@1650898796.887301189","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3134,"referenced_errors":[{"created":"@1650898796.887300282","description":"failed to connect to all addresses","file":"src/core/lib/transport/error_utils.cc","file_line":163,"grpc_status":14}]}"
>

During handling of the above exception, another exception occurred:

ConnectionFailedError                     Traceback (most recent call last)
<ipython-input-1-8e27a66a0b84> in <module>()
      1 from lwa352_pipeline_control import Lwa352CorrelatorControl
----> 2 p = Lwa352CorrelatorControl( ['lxdlwagpu0%d' % i for i in range(1,9) ], npipeline_per_host=4)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in __init__(self, hosts, npipeline_per_host, etcdhost, log)
     78                                                             pipeline_id=pipeline_id,
     79                                                             etcdhost=etcdhost,
---> 80                                                             log=log,
     81                                                         )]
     82         self.npipeline = len(self.pipelines)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in __init__(self, host, pipeline_id, etcdhost, log)
    344         self.beamform_output = BeamformOutputControl(*args)
    345         self.beamform_vlbi_output = BeamformVlbiOutputControl(*args)
--> 346         if not self.check_connection():
    347             raise RuntimeError("Connection failed. Consider restarting lwa-xeng-etcd.service daemon on host %s" % host)
    348 

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in check_connection(self, timeout)
    374         """
    375         try:
--> 376             self.corr_interface.send_command(self.host, cmd='get_pid', block='xctrl', xid=self.pipeline_id, timeout=timeout)
    377             return True
    378         except RuntimeError:

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/etcd_control.py in send_command(self, host, pipeline, block, inst_id, cmd, timeout, **kwargs)
    267 
    268         # Begin watching response channel and then send message
--> 269         watch_id = self.ec.add_watch_callback(resp_key, response_callback)
    270         # send command
    271         self.ec.put(cmd_key, command_json)

~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py in handler(*args, **kwargs)
     46                 return f(*args, **kwargs)
     47             except grpc.RpcError as exc:
---> 48                 _translate_exception(exc)
     49 
     50     return functools.wraps(f)(handler)

~/anaconda3/envs/casa/lib/python3.6/site-packages/etcd3/client.py in _translate_exception(exc)
     30     if exception is None:
     31         raise
---> 32     raise exception
     33 
     34 

ConnectionFailedError: etcd connection failed

rh-codebase, 2022-04-25:
mcs host is denying logins so it's in a bad state. etcd server runs on the mcs host.

rh-codebase, 2022-04-27:
I got onto the IPMI interface and the screen is filled with:
Write-error on swap-device
and
EXT4-fs error (device nvme0n1p2)
Not good signs and may be related to those power glitches and the drive suffering hidden problems. The system rebooted and things look to be back up. Keeping issue open in case this happens again in which case a new nvme disk should be ordered.

rh-codebase, 2022-07-07:
closing since this appears to be a one time occurrence.

closed 2022-07-07 by rh-codebase