ovro-lwa/lwa-issues #111: swdeploy service using 100% CPU on all GPU nodes

rh-codebase, 2022-04-28:
service restarted by jack. No other info available. Maybe related to when MCS was rebooted. Check grafana to see if gpu system load can be correlated with MCS reboot.

rh-codebase, 2022-04-28:
[image]
[image]
There does look to be correlation between GPU CPU load and lack of network traffic. The MCS was restarted on the right side of MCS graph. The lack of traffic could be the result of Etcd being down although during the large time MCS was down, there looks to be a gradual increase in CPU load. Kind of strange. The linux host's Prometheus client seem to have some buffering as data shows up even when MCS was down. This should be easily tested by shutting down Etcd next maintenance day.

jack-h, 2022-08-03:
This is pinned at 100% CPU use again. I won't restart the service unless instructed in case there is some useful debugging to be done with it still live

caseyjlaw, 2023-08-17:
I'm still seeing 100% CPU for this service on the gpu servers.

rh-codebase, 2023-08-17:
yep. restarted on gpu01-08. This process sure doesn't like having etcd pulled out from beneath it. Will look into it....after dsa2k PDR. :)