ovro-lwa/lwa-issues #115: No dr-vfast service on gpu09

marinanderson, 2022-05-11:
I see only dr-beam, dr-vslow, and dr-tengine services on lxdlwagpu09?

marinanderson, 2022-05-11:
Renamed this issue so it is applicable for general dr-vfast problems, until the mode becomes operational.

dr-vfast-* worked (thanks, Jayce!), and all 16 fast recorders were found by Lwa352RecorderControl. However, from the log (output below is from dr-vfast-1), it looks like the digital system isn't sending any data to the recorders?

> 2022-05-11 18:45:34 [DEBUG   ] === Performance Report ===
2022-05-11 18:45:34 [DEBUG   ]  max acquire/process/reserve times: 0.000/0.000/0.000
2022-05-11 18:45:34 [DEBUG   ]  pipeline lag: 0.0
2022-05-11 18:45:34 [DEBUG   ]  load average: 0.23, 0.35, 1.26
2022-05-11 18:45:34 [DEBUG   ] ===   ===
2022-05-11 18:45:34 [DEBUG   ] === Storage Report ===
2022-05-11 18:45:34 [DEBUG   ]  directory: /home/ubuntu/data/fast
2022-05-11 18:45:34 [DEBUG   ]  disk size: 983428714496 B
2022-05-11 18:45:34 [DEBUG   ]  disk free: 155119026176 B
2022-05-11 18:45:34 [DEBUG   ]  file count: 0
2022-05-11 18:45:34 [DEBUG   ]  total size: 0 B
2022-05-11 18:45:34 [DEBUG   ] ===   ===
2022-05-11 18:45:34 [DEBUG   ] === Status Report ===
2022-05-11 18:45:34 [DEBUG   ]  queue size: 1
2022-05-11 18:45:34 [DEBUG   ]  active operation: True
2022-05-11 18:45:34 [DEBUG   ]  active filename: fast
2022-05-11 18:45:34 [DEBUG   ]  active time remaining: 364 days, 23:47:33.312040

jaycedowell, 2022-06-29:
A pipeline lag of zero would indicate no data or that all of the packets are getting drop by the receiver.  What address/port pairs are the fast data going to?

jaycedowell, 2022-06-30:
I took a closer look at this today and I was able to record something.  I think the missing piece for the fast visibilities is configuring the correlator to send packets.  What I did was a:
```
from lwa352_pipeline_control import Lwa352CorrelatorControl 

p = Lwa352CorrelatorControl( ['lxdlwagpu0%'% i for i in range(1,9) ], npipeline_per_host=4)

// Bring up the pipelines
p.stop_pipelines()
p.start_pipelines()

// Get slow flowing
 p.configure_corr(dest_ip=['10.41.0.25', '10.41.0.25', '10.41.0.73', '10.41.0.73'], dest_port=[10001+i//4 for i in range(32)] )

// Get fast flowing
dest_ip=['10.41.0.25', '10.41.0.25', '10.41.0.73', '10.41.0.73']
dest_port = [11001+i//4 for i in range(32)]
for i in range(32):
    p.pipelines[i].corr_output_part.set_destination(dest_ip[i%4], dest_port[i])
```

If you want to look at what I recorded there are a collection of files in `/home/ubuntu/data/fast` on gpu09.  They should each have 10 100 ms integrations in them.

closed 2022-08-04 by jaycedowell