ovro-lwa/lwa-issues #116: 1s discrepancy between LWA-SV and OVRO LWA

iveydavis, 2022-05-11:
The moon bounce showed there's an exactly 1 second time discrepancy between signals received by the LWA-SV station and the LWA at OVRO; the OVRO LWA is reporting the signal 1s later than it should be.

iveydavis, 2022-05-17:
sorry for completely missing the questions at the end of this. How I understand it is that they took the unix timestamp from the hdf5 files and compared it to the estimated arrival time for the chirp-- SV was on time and OVRO was off by that 1s. I don't think there was any integration of the data from the hdf5 files.

jaycedowell, 2022-10-21:
I'm going to try to look at this today in the moon bounce data using WWV.

jaycedowell, 2022-10-21:
I'm looking at data from 10/20, specifically the 059872_173000000000c6e0d48 file from yesterday's moon bounce experiment.  I'm also using [this page on the contents of WWV and WWVH](https://www.nist.gov/time-distribution/radio-station-wwv/wwv-and-wwvh-digital-time-code-and-broadcast-format) as a reference.  The easiest thing to do is look for what that page calls the "standard time intervals" which are ticks on the second.

To process the data I take the raw file and convert it to spectra using `hdfWaterfall.py` with 16,384 channels and ~2 ms time resolution.  I think down select those data to 50 kHz centered on the 10 MHz WWV single.  After that I then sum XX and YY to make psuedo-Stokes I and then only keep the two channels closest to +/-1 kHz of the carrier.  When you do that you get something like this:

[image]

The wide peak at 115 is the 800 ms "minute marker" and you can see the 1 s ticks after that.  Zooming into the minute marker it really does look like it is 800 ms wide:

[image]

If I pull out the timestamp for that first "x" in the initial rise of the marker I get that it is at 17:32:00.004.  Based on that I would say that there is no 1 s offset.  I'll try to LWA-SV data next.

jaycedowell, 2022-10-21:
For the LWA-SV file 059872_004750344 I follow the same process and see:

[image]

That's the same minute marker and it starts at 17:32:00.003.  For at least this one snapshot everything looks ok.

iveydavis, 2022-10-21:
That's great to hear that it doesn't look like there's a time discrepancy now! I don't know if this is because there _was_ a problem that's now resolved, or if there was an error in Mark's analysis of the original observation. I'd be interested to see his analysis of this data too.

jaycedowell, 2022-10-22:
Yeah, it's not clear.  It could also be that there is a discrepancy in the time tagging in the power beam data that were used previously.  I did collect power beam data for these observations so that might be something to look at.  Maybe the second ticks can be pulled out of that even though the spectral resolution is much lower.

jaycedowell, 2022-10-28:
In light of what I'm seeing with #189 it could be both that the time keeping in the F-engines/X-engines/T-engine is ok but the time keeping in the power beam recorders is not.

caseyjlaw, 2023-12-14:
Resolved with new commit, as described in #189

closed 2023-12-14 by caseyjlaw