ovro-lwa/lwa-issues #117: GPU09 crashed while collecting solar beamforming data

rh-codebase, 2022-05-16:
Not much known about this so a placeholder to track future issues on this topic. Before the crash event on 2022-05-16T19:10ish UT, gpu09 had a system load in the 20's with high interrupts and context switching and using about 350Watts. Temperatures rose a bit but not too much. See Grafana during this time. Similar levels were seen on 2022-05-13T17:40 UT but w/out issue. Solar data collection previously on 2022-05-12T17:10 also had no issues. Nothing stands out in logs. The crash resulted in an attempted reboot. The console screen(via IPMI) was stuck during the BIOS 'Initializing ....' phase. The host booted w/out issue after a remote powercycle.

caseyjlaw, 2022-05-17:
I just killed gpu09 while collecting slow visibility data. Observation started around 6am Tuesday and crashed around 6:30am.

rh-codebase, 2022-05-17:
restarted gpu09. Started 20 cpu intensive tasks(infinite loop c-code) at 2022-05-17T16:41:40(see grafana). System load in the 20's. Temp leveled off around 45C. Power around 350W. Running for about 7min and no issues. Will let it run for another 10min.

yupinghuang, 2022-05-17:
syslog entry?
```
May 17 06:03:13 127.0.0.1 ^[[2m2022-05-17T13:03:13 lxdlwagpu09 ./py[]: ^[[0m ^[[1m{"mjd": 59716.54390752178, "proj": "LWA", "subsystem": "cmd_rsp", "app": "_", "version": "_", "module": "CmdRsp", "function": "c-tor", "level": "info", "time": "2022-05-17T13:03:13.609875Z", "msg": "Created CmdRsp object"}^[[0m
May 17 06:03:13 127.0.0.1 ^[[2m2022-05-17T13:03:13 lxdlwagpu09 ./py[]: ^[[0m ^[[1m{"mjd": 59716.5439075295, "proj": "lwa", "subsystem": "arx", "app": "_", "version": "_", "module": "Arx", "function": "c-tor", "level": "info", "time": "2022-05-17T13:03:13.610543Z", "msg": "Created Arx object"}^[[0m
May 17 09:37:49 127.0.0.1 /home/ubuntu/proj/lwa-shell/goshutdown/cmd/goshutdown/goshutdown[2811]: {"level":"info","msg":"Version: v1.5.1, Build: bcf617b4e992ba5499756a1a08f48ab94aa2303e","time":"2022-05-17T09:37:49-07:00"}
```
But the way. There's nothing in /var/log that looks like logs from the OS???

caseyjlaw, 2022-05-17:
Not sure if this is related, but I'm able to crash the server (or maybe just the ssh server?) while tarring and untarring files in ~/data/slow.

The slow data is recorded in MS format and tarred. I created a function to untar, read, and tar it back again:
```
os.system(f"tar xvf {filename}")
tb = casacore.tables.table(filename.rstrip('.tar'))
os.system(f"tar cvf {filename} {filename.rstrip('.tar')}")
```

If I run this as a function on tens of files, then it tends to cause the crash. Before crashing, it also raises os warnings about some files not being interpreted as tar files. 
After the crash, I can wait about 5 minutes and ssh access is restored.

rh-codebase, 2022-05-18:
the 'uptime' command shows the total time the system has been up since last reboot. This should show whether the entire system rebooted. systemctl status ssh should show the uptime of the ssh service.

rh-codebase, 2022-05-18:
Can someone see if any of this is reproducible on another gpu host?

rh-codebase, 2022-05-18:
I ran the following a few times on gpu09 w/out any issues:
tar cf test.tar.gz proj
mkdir junk
cd junk
tar xf ../test.tar.gz
rm -rf proj
cd ..
rm test.tar.gz
.... repeat the above a few times, about 4 with no issues. proj is about 4.3GB in size with a good number of files.

yupinghuang, 2022-05-18:
ok the fact that it only shows up on python reminds me of this https://stackoverflow.com/questions/1180606/using-subprocess-popen-for-process-with-large-output

yupinghuang, 2022-05-18:
Casey and I ran the pipeline a handful of times this morning and reliably reproduced the issue. stdout buffer doesn't seem to be an issue. /dev/shm was not overflowing (and Jayce made the change not to use it). The only thing I noticed was that the buffer/cache memory usage and the actual memory usage of the data recorders seemed to creep up by 1 GB per ~20s but the node crashes when there are still ~500 GB of RAM left (at 2s resolution). So it seems like the software wasn't directly causing the OS to crash.

I don't know if it's common for the buffer/cache to creep up when data recorders are running, but maybe there's a disk issue with the NVMe. I'll run the pipeline again while watching dmesg and iotop and maybe run an fsck on /home and see what happens. I'll also figure out how to write the kernel log to files...

Rick, let us know if you have other suspicions.

caseyjlaw, 2022-05-18:
I think the disk issue is worth investigating more.
I ran more tests in which I read files from disk for slow visibilities recorded on 2022-05-17 and 2022-05-18. Files from the 18th can be read without issue. Reading files from the 17th reliably crashes gpu9. It also crashes if I control the stderr stream from the Python command execution (a potential issue raised by Yuping). The number of files on the 17th is about 3x higher, but the crash happens quickly, so I don't think it is related to the number of files.
A bad disk would also help explain why gpu9 crashes in all the scenarios we've seen:
- reading files
- data recorder on /dev/shm
- data recorder in ~/data/slow

yupinghuang, 2022-05-19:
Another option would be to reprovision the node with the OS on a different NVMe and then fsck this NVMe...

yupinghuang, 2022-05-19:
`dmesg` just gave me this. Don't know if it's a red herring but `vendor_id: 0x144d` is Samsung, but I don't know which one of the NVMes the message is referring to. I'll work on configuring syslog to write kernel log to disk next.

```
[11189.970968] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757589 offset:0x140 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:180)
[11219.196524] {1}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
[11219.196558] {1}[Hardware Error]: event severity: recoverable
[11219.196576] {1}[Hardware Error]:  Error 0, type: recoverable
[11219.196591] {1}[Hardware Error]:   section_type: PCIe error
[11219.196606] {1}[Hardware Error]:   port_type: 0, PCIe end point
[11219.196622] {1}[Hardware Error]:   version: 3.0
[11219.196636] {1}[Hardware Error]:   command: 0x0546, status: 0xc110
[11219.196653] {1}[Hardware Error]:   device_id: 0000:5e:00.0
[11219.196668] {1}[Hardware Error]:   slot: 9
[11219.196673] mce: [Hardware Error]: Machine check events logged
[11219.196680] {1}[Hardware Error]:   secondary_bus: 0x00
[11219.196684] {1}[Hardware Error]:   vendor_id: 0x144d, device_id: 0xa808
[11219.196715] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[11219.196726] {1}[Hardware Error]:   class_code: 010802
[11219.196745] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc00188001010092
[11219.196746] EDAC skx MC3: TSC 18c66e86f378 
[11219.196746] EDAC skx MC3: ADDR b7575807c0 
[11219.196747] EDAC skx MC3: MISC 200800c030002086 
[11219.196748] EDAC skx MC3: PROCESSOR 0:50657 TIME 1652920975 SOCKET 1 APIC 20
[11219.196755] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757580 offset:0x7c0 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:18)
[11219.196807] nvme 0000:5e:00.0: aer_status: 0x00001000, aer_mask: 0x00100000
[11219.196830] Poisoned TLP
[11219.196832] nvme 0000:5e:00.0: aer_layer=Transaction Layer, aer_agent=Receiver ID
[11219.196853] nvme 0000:5e:00.0: aer_uncor_severity: 0x00062030
[11219.196870] nvme 0000:5e:00.0:   TLP Header: 4a004040 00000100 5e001400 60c4eb57
```

rh-codebase, 2022-05-19:
looks like 2 diff. errors going on. The EDAC is usually referring to ECC RAM and then there's the nvme lines. This may be worth reading:

https://blog.ioncomputer.com/2019/05/06/linux-edac-modules-on-server-systems/

and TLP:

https://www.intel.com/content/www/us/en/docs/programmable/683514/16-1/transaction-layer-errors.html

and

https://nvmexpress.org/wp-content/uploads/NVMe-PCIe-Transport-Specification-1.0a-2021.07.26-Ratified.pdf

This may not be the nvme but a PCI error??

yupinghuang, 2022-05-19:
The NVMes are seated on a PCIe slot, though. Kathryn noted that she had to made a custom bracket for the nvme adapter and maybe maybe it needs a reseating.

rh-codebase, 2022-05-19:
worth a try. reassigning

yupinghuang, 2022-05-20:
ok I just saw this in `dmesg -wH` right before the node freezes. So not the NVMes. Apparently the MCEs can come from either the CPUs or the RAM sticks and [reddit ](https://www.reddit.com/r/homelab/comments/da3bp8/getting_a_lot_of_corrected_ecc_errors/)tells me that the ipmi console may log which device is causing the issue.

Rick, could me let me know if there's a way I can access the node's ipmi? Or if you can figure out which device is causing us trouble from the log?

I...thought I remember the login to maas but I actually don't so I'll wait for the node to come up to try again just to be sure that the same errors get logged. Meanwhile, I don't think reseating the NVMe expansion card is gonna help.

```
[May19 19:57] CMCI storm detected: switching to poll mode
[  +0.000127] mce: [Hardware Error]: Machine check events logged
[  +0.000013] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[  +0.000001] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc00010001010092
[  +0.000001] EDAC skx MC3: TSC e6ee62e90fe6 
[  +0.000001] EDAC skx MC3: ADDR b757589dc0 
[  +0.000000] EDAC skx MC3: MISC 200805c290402086 
[  +0.000002] EDAC skx MC3: PROCESSOR 0:50657 TIME 1653015421 SOCKET 1 APIC 20
[  +0.000006] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757589 offset:0xdc0 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:1a8)
[  +0.000002] mce: [Hardware Error]: Machine check events logged
[  +0.000004] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[  +0.000001] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc0001c001010092
[  +0.000000] EDAC skx MC3: TSC e6ee62eaaf16 
[  +0.000001] EDAC skx MC3: ADDR b757599940 
[  +0.000000] EDAC skx MC3: MISC 200805c297a02086 
[  +0.000001] EDAC skx MC3: PROCESSOR 0:50657 TIME 1653015421 SOCKET 1 APIC 20
[  +0.000003] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757599 offset:0x940 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:240)
[  +0.751069] mce: Uncorrected hardware memory error in user-access at b757585bc0
[  +0.071262] Memory failure: 0xb757585: Killing python3:10967 due to hardware memory corruption
[  +0.000068] Memory failure: 0xb757585: recovery action for dirty LRU page: Recovered
[  +0.326683] ghes_edac: Internal error: Can't find EDAC structure
[  +0.000032] {1}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
[  +0.000023] {1}[Hardware Error]: event severity: recoverable
[  +0.000016] {1}[Hardware Error]:  Error 0, type: recoverable
[  +0.000015] {1}[Hardware Error]:   section_type: memory error
[  +0.000016] {1}[Hardware Error]:   node: 3 module: 0 
[  +0.000015] {1}[Hardware Error]:   DIMM location: not present. DMI handle: 0x0000 
[  +0.000019] {1}[Hardware Error]:  Error 1, type: recoverable
[  +0.000015] {1}[Hardware Error]:   section_type: general processor error
[  +0.000017] {1}[Hardware Error]:   processor_type: 0, IA32/X64
[  +0.000015] {1}[Hardware Error]:   processor_isa: 2, X64
[  +0.000014] {1}[Hardware Error]:   error_type: 0x01
[  +0.000014] {1}[Hardware Error]:   cache error
[  +0.000014] {1}[Hardware Error]:   operation: 0, unknown or generic
[  +0.000016] {1}[Hardware Error]:   version_info: 0x0000000000050657
[  +0.000017] {1}[Hardware Error]:   processor_id: 0x0000000000000036
[  +0.594565] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[  +0.000003] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc000e0001010092
[  +0.000000] EDAC skx MC3: TSC 0 
[  +0.000001] EDAC skx MC3: ADDR b757585bc0 
[  +0.000000] EDAC skx MC3: MISC 200800c030002086 
[  +0.000002] EDAC skx MC3: PROCESSOR 0:50657 TIME 1653015423 SOCKET 1 APIC 20
[  +0.000007] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757585 offset:0xbc0 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:f8)
[  +4.991982] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[  +0.000003] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc00050001010092
[  +0.000001] EDAC skx MC3: TSC 0 
[  +0.000001] EDAC skx MC3: ADDR b757599940 
[  +0.000001] EDAC skx MC3: MISC 200805c2b0002086 
[  +0.000001] EDAC skx MC3: PROCESSOR 0:50657 TIME 1653015428 SOCKET 1 APIC 20
[  +0.000008] EDAC MC3: 1 UE memory read error on CPU_SrcID#1_MC#1_Chan#2_DIMM#0 (channel:2 slot:0 page:0xb757599 offset:0x940 grain:32 -  OVERFLOW err_code:0101:0092 socket:1 imc:1 rank:0 bg:3 ba:0 row:1a1e8 col:240)
[  +9.339677] mce: Uncorrected hardware memory error in user-access at b7575831c0
[  +0.102813] Memory failure: 0xb757583: Killing python3:10773 due to hardware memory corruption
[  +0.000069] Memory failure: 0xb757583: recovery action for dirty LRU page: Recovered
[  +0.834294] ghes_edac: Internal error: Can't find EDAC structure
[  +0.000033] {2}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
[  +0.000014] mce: Uncorrected hardware memory error in user-access at b757587fc0
[  +0.000014] {2}[Hardware Error]: event severity: recoverable
[  +0.000007] EDAC skx MC3: HANDLING MCE MEMORY ERROR
[  +0.000016] {2}[Hardware Error]:  Error 0, type: recoverable
[  +0.000026] EDAC skx MC3: CPU 10: Machine Check Event: 0 Bank 8: fc000c4001010092
[  +0.000003] EDAC skx MC3: TSC 0 
[  +0.000001] EDAC skx MC3: ADDR b7575831c0 
[  +0.000018] {2}[Hardware Error]:   section_type: memory error
[  +0.000001] EDAC skx MC3: MISC 200800c037e02086 
[... cut ...]
[  +0.000018] {2}[Hardware Error]:   processor_id: 0x0000000000000004
[  +0.015103] Memory failure: 0xb757587: Killing python3:10800 due to hardware memory corruption
[  +0.000042] Memory failure: 0xb757587: recovery action for dirty LRU page: Recovered
[  +0.315365] mce: Uncorrected hardware memory error in user-access at b757589740
[  +0.008035] Memory failure: 0xb757589: Killing python3:10883 due to hardware memory corruption
[  +0.001279] Memory failure: 0xb757589: recovery action for dirty LRU page: Recovered
```

yupinghuang, 2022-05-20:
Unless any of you all have better ideas, I'm gonna go on a limp and take `MC#1_Chan#2_DIMM#0` which is present in all the memory errors...and...claim that it is `P0_Node1_Channel2_Dimm0`, since there are 4 mcs and 3 dimms per mc. This is `P1-DIMMF1` on the motherboard schematic. The other channel2_dimm0s are P2-DIMMC1, and P2-DIMMF1 (page 40 of the motherboard manual shows where these slots are https://www.supermicro.com/en/products/motherboard/x11dgq).

So we either pull our best suspect out, reboot the system, and see if things work. Or we pull all three out first.

refs
https://support.siliconmechanics.com/portal/en/kb/articles/identify-bad-dimm-from-edac
https://www.supportsages.com/dimm-errors/

```
(casa) ubuntu@lxdlwagpu08:~$ ls -s /sys/devices/system/edac/mc/
total 0
0 mc0  0 mc1  0 mc2  0 mc3  0 power  0 subsystem  0 uevent
(casa) ubuntu@lxdlwagpu08:~$ ls -s /sys/devices/system/edac/mc/mc1
total 0
0 ce_count  0 ce_noinfo_count  0 dimm0  0 dimm2  0 dimm4  0 max_location  0 mc_name  0 power  0 reset_counters  0 seconds_since_reset  0 size_mb  0 subsystem  0 ue_count  0 ue_noinfo_count  0 uevent
```

```
(casa) ubuntu@lxdlwagpu08:~$ sudo dmidecode -t 17
# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.
# SMBIOS implementations newer than version 3.1.1 are not
# fully supported by this version of dmidecode.

Handle 0x0013, DMI type 17, 84 bytes
Memory Device
	Array Handle: 0x0012
	Error Information Handle: Not Provided
	Total Width: 72 bits
	Data Width: 64 bits
	Size: 64 GB
	Form Factor: DIMM
	Set: None
	Locator: P1-DIMMA1
	Bank Locator: P0_Node0_Channel0_Dimm0
	Type: DDR4
	Type Detail: Synchronous
	Speed: 2933 MT/s
	Manufacturer: Samsung
	Serial Number: 23D9466C
	Asset Tag: P1-DIMMA1_AssetTag (Date:20/39)
	Part Number: M393A8G40MB2-CVF    
	Rank: 2
	Configured Clock Speed: 2400 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V

Handle 0x0014, DMI type 17, 84 bytes
Memory Device
	Array Handle: 0x0012
	Error Information Handle: Not Provided
	Total Width: 72 bits
	Data Width: 64 bits
	Size: 64 GB
	Form Factor: DIMM
	Set: None
	Locator: P1-DIMMB1
	Bank Locator: P0_Node0_Channel1_Dimm0
	Type: DDR4
	Type Detail: Synchronous
	Speed: 2933 MT/s
	Manufacturer: Samsung
	Serial Number: 23D94710
	Asset Tag: P1-DIMMB1_AssetTag (Date:20/39)
	Part Number: M393A8G40MB2-CVF    
	Rank: 2
	Configured Clock Speed: 2400 MT/s
	Minimum Voltage: 1.2 V
[... 218 lines cut ...]

Handle 0x001E, DMI type 17, 84 bytes
Memory Device
	Array Handle: 0x0012
	Error Information Handle: Not Provided
	Total Width: 72 bits
	Data Width: 64 bits
	Size: 64 GB
	Form Factor: DIMM
	Set: None
	Locator: P2-DIMMF1
	Bank Locator: P1_Node1_Channel2_Dimm0
	Type: DDR4
	Type Detail: Synchronous
	Speed: 2933 MT/s
	Manufacturer: Samsung
	Serial Number: 23D946FE
	Asset Tag: P2-DIMMF1_AssetTag (Date:20/39)
	Part Number: M393A8G40MB2-CVF    
	Rank: 2
	Configured Clock Speed: 2400 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
```

rh-codebase, 2022-05-20:
Finding IPMI IP for hosts under MaaS control:
Login to MaaS
Select the desired host from the list.
Select the 'Configuration' Menu (upper right)
In the 'Power Configuration' section, the IP is shown.
Setup a tunnel to that IP on port 443.
url should then be:  https://localhost:<local_tunnel_port>

yupinghuang, 2022-05-20:
Ha! Got it! 
```
Uncorrectable ECC / other uncorrectable memory error @P2-DIMMF1 - Assertion
```

From ipmi -> server health -> health event log.

I'd suggest that we remove the stick, turn on the node and see if it still causes troubles, then try reseating it. It may well be that the RAM stick needs a replacement.

yupinghuang, 2022-05-20:
From Jack:

I don't think we have spares, but GPU09 has lots of memory in (needed by the correlator, but not really by the recorder), so just pulling /disabling the bad stick/(or even the whole channel if that's necessary) shouldn't cause any problems. (This is where Jayce tells me that's not true)

rh-codebase, 2022-05-20:
memory stick: p2-dimmF1 has been removed from gpu09 and the system is up for testing.

caseyjlaw, 2022-05-20:
I've rerun the test that triggered the failure and I no longer see gpu9 freeze up.

rh-codebase, 2022-05-20:
It could be that the module just needs to be re-seated. Do you feel like testing this out?

rh-codebase, 2022-05-25:
module re-seat failed. Will order new one.

closed 2022-05-25 by rh-codebase