ovro-lwa/lwa-issues #123: slow data recorder resiliency

caseyjlaw, 2022-05-25:
I will admit this is not reproducible, but I wanted to set a placeholder for some issues related to the reliability of the slow data recorder.
I am working with in a jupyter notebook and have a pattern of using the mnc.control interface to start subsystems in sequence. Then I want to record slow visibilities while walking through a fake "schedule" that modifies the digital configuration (e.g., set inputs to zeros and check ms files).
Sometimes ms files do not show up. I have one error in the slow data recorder pasted here:

```
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: Error in atexit._run_exitfuncs:
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: Traceback (most recent call last):
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/shutil.py", line 475, in rmtree
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]:     orig_st = os.lstat(path)
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/MeasurementSetWriter-28001'
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: Error in atexit._run_exitfuncs:
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: Traceback (most recent call last):
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/shutil.py", line 475, in rmtree
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]:     orig_st = os.lstat(path)
May 25 12:22:40 lxdlwagpu09 dr-vslow-1[28001]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/MeasurementSetWriter-28001'
```

I saw another error related to threading. Neither have been reproduced, but perhaps this triggers some ideas for improving resiliency.

jaycedowell, 2022-05-25:
With your schedule, how fast were you zeroing inputs, starting a recording, and then checking for files?

caseyjlaw, 2022-05-25:
I would set everything up. Confirm that ms files were being written. Then use f-engine controls to change inputs every 30 seconds.

jaycedowell, 2022-05-25:
30 seconds isn't all that fast but is is comparable to the last set of pipeline lags I see reported in the logs.  How often do you end up with missing ms files and are the missing files across all subbands or only a few?

caseyjlaw, 2022-05-25:
Ah, I should have made that more clear. I am trying to track the problem where no ms files are written, despite the python-level controls saying that things are fine.
My tests assume that f-engine inputs can be modified freely without concern to data recorders. The candence of changes was set slow enough so that I can identify each state clearly in the 10 second ms files.

marinanderson, 2022-05-28:
After some investigations by Kathryn and myself, we discovered that we were repeatedly seeing no MS files written because the disk on gpu09 was filling up.

No errors were returned until running `rs.stop()`, which reported 
```
The history saving thread hit an unexpected error (OperationalError('database or disk is full',)).History will not be written to the database.
ERROR1: timeout
```

I see that the recorder now writes `.ms` files instead of `.ms.tar` files. Is the buffer that runs on `dr-vslow-1` still looking for `.ms.tar` files to delete by any chance? A quick scan through the recent `dr-vslow-1` log files show no more of the `=== Quota Report ===` summaries on the number of files removed.

jaycedowell, 2022-05-31:
Can you post the quota management problem as an issue at ovro_data_recorder?

caseyjlaw, 2022-05-31:
I'll do that.
Just a note that the original issue posted here is distinct, but not reproducible yet.

closed 2022-11-02 by caseyjlaw