ovro-lwa/lwa-issues #124: GPU servers unreachable

kathrynplant, 2022-05-27:
The gpu servers aren't responding to pings from lwamaas.

rh-codebase, 2022-05-27:
thermal shutdown. bringing hosts back up and retuning.

caseyjlaw, 2022-05-27:
Ooh, a real world test. Do you think it worked as intended?
Looking at the Grafana "LWA Shutdown" dashboard, I see that there is a "stale temp" state change. If that was the trigger, then the shutdown was caused by a lack of measurement, rather than high temp. Is that the right interpretation?

rh-codebase, 2022-05-27:
new setpoint at 24.5C

closed 2022-05-27 by rh-codebase

rh-codebase, 2022-10-11:
not sure what you are looking at but the shutdown was indeed due to the setpoint being reached. There is some tuning to do in regards to the shutdown message being sent before the machine shuts down.