ovro-lwa/lwa-issues #134: lwacalim02 missing interface?

caseyjlaw, 2022-07-20:
Working on data capture on Cal-Im, I noticed that `lwacalim02` is different from the others.
To wit:
```
> ip -h address
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp68s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 3c:ec:ef:61:37:6a brd ff:ff:ff:ff:ff:ff
    inet 10.40.0.211/24 brd 10.40.0.255 scope global dynamic noprefixroute enp68s0f0
       valid_lft 155259583sec preferred_lft 155259583sec
    inet6 fe80::3eec:efff:fe61:376a/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: enp68s0f1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 3c:ec:ef:61:37:6b brd ff:ff:ff:ff:ff:ff
4: virbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 52:54:00:1c:9d:58 brd ff:ff:ff:ff:ff:ff
    inet 192.168.122.1/24 brd 192.168.122.255 scope global virbr0
       valid_lft forever preferred_lft forever
```

Should it have an interface in the `10.40.0.7x` range?

rh-codebase, 2022-07-20:
the 40Gbe card is not on the pci bus anymore:

(base) [ubuntu@lwacalim02 ~]$ lspci | grep -i ether
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)

There should be LX719 40Gbe

Will have to power down machine a try reseating the card.

rh-codebase, 2022-07-21:
40G card reseated. Also needed to reinsert fiber to switch. Interface is up.

rh-codebase, 2022-08-26:
interface not visible again. First try reseating the fiber connection on switch and if that doesn't bring it up, then reseat the card.

rh-codebase, 2022-09-03:
calim02 40G card back online.

closed 2022-09-03 by rh-codebase