ovro-lwa/lwa-issues #135: Data recorder deployment on lwacalim

marinanderson, 2022-07-21:
Could not get any data to flow when testing out the data recorder on the lwacalim nodes. (Quick test of data recorder on gpu09 showed things were working find over there.)

After starting up the GPU pipelines, I ran the following:
`p.configure_corr(dest_ip=['10.41.0.%d' % s for s in range(76,84) for i in range(4)], dest_port=[10001+i//2 for i in range(4)]*8 )`
for IPs 76 through 84, corresponding to lwacalim01 through 08.

As `pipeline` user on `lwacalim01`:
`python3 ~/ovro_data_recorder/dr_visibilities.py --address 10.41.0.76 --record-direcgtory ~/fast/test --port 10001 --cores 50,51,52,53,54,55 --record-directory-quota 50GB --no-tar`

rh-codebase, 2022-07-21:
did you really specify:  --record-direcgtory?  If so, then the code should have thrown some sort of unknown argument error.

marinanderson, 2022-07-21:
Nope, errant g when filing the issue. The actual command didn't have the typo.

jaycedowell, 2022-07-22:
I think the MTU on that link needs be to be upped.

rh-codebase, 2022-07-22:
indeed. I'll reassign back to jayce when the MTU is changed.

rh-codebase, 2022-07-22:
mtu set to 9000 on lwacalim00 - 09

jaycedowell, 2022-07-22:
Is there a plan to bring up a second 40G interface on the calim nodes?  If not that link may need to have its netmask updated so it can see all of the x-engines.

jaycedowell, 2022-07-22:
A few things:
 1) There is a firewall on the calim nodes that is dropping the UDP packets from the x-engines.  I took me a long time to figure that out and I have added a couple of exceptions on lwacalim01 but something more permanent needs to be done.
 2) I needed to update the CASA data that lives in the conda "datarecorder" environment.  This was more of a one-time thing.
 3) For what ever reason I get a password prompt for when trying to log in as pipeline on lwacalim02.  The other nodes seem ok.
 4) I've started on a new "calim" branch in ovro_data_recorder that should build the correct systemd services for calim.

yupinghuang, 2022-07-22:
1. I'll configure the firewall on all nodes to allow for 10001/udp and 10002/udp traffic (or whatever is in the service file). Sorry I should have mentioned that being a potential issue (running `firewall-cmd` with the `--permanent` option makes it...permanent.
3. It seemed like the NIS client didn't come up after the ethernet card was reseated. Reran `sudo authconfig --enablenis --nisdomain=calim --nisserver=10.40.0.206 --update` and it seems fine now.

yupinghuang, 2022-07-22:
> Is there a plan to bring up a second 40G interface on the calim nodes? If not that link may need to have its netmask updated so it can see all of the x-engines.

There is no plan to bring up a second 40G interface.

jaycedowell, 2022-07-22:
The ports would be 10001/2 for slow, 11001/2 for fast, and 20001/2 for the power beams.

yupinghuang, 2022-07-22:
Done
```
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=10001/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=10002/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=11002/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=11001/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=20001/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --permanent --add-port=20002/udp'
pdsh -w lwacalim[00-10] 'sudo firewall-cmd --reload'
```

jaycedowell, 2022-07-22:
Super, thanks.

jaycedowell, 2022-07-27:
### Services
The services are setup on calim.  I've added a couple of helper shell scripts to manage them to ovro_data_recorder/services.  There is:
 * install_dr_services.sh - Copies the service files to /etc/systemd/system using the hostname to figure out what goes where
 * start_dr_services.sh - Starts _all_ services on a node
 * restrart_dr_services.sh - Restarts _running_ services on a node
 * stop_dr_services.sh - Stops _running_ services on a node.

I found them helpful as an easier way to start/stop the collection of services.

### Setup
For getting the slow and fast visibilities running I've been using:
```
from lwa352_pipeline_control import Lwa352CorrelatorControl 

p = Lwa352CorrelatorControl(['lxdlwagpu0%d' % i for i in range(1,9)],
                                                     npipeline_per_host=4)

// Slow
p.configure_corr(dest_ip=['10.41.0.%d' % s for s in range(76,84) for i in range(4)],
                             dest_port=[10001, 10001, 10002, 10002])

// Fast
dest_ip=['10.41.0.%d' % s for s in range(76,84) for i in range(4)]
dest_port = [11001, 11001, 11002, 11002]
for i in range(32):
    p.pipelines[i].corr_output_part.set_destination(dest_ip[i], dest_port[i%4])
```
On the data recorder side I currently have data going to /data## on the calim nodes.  I've set a fairly low quota size on each node of 500 GB for slow visibilities and 250 GB for fast.

### Control
I've been having trouble with the `mnc.ezdr.Lwa352RecorderControl`.  It throws a "Received message larger than max (4394477 vs. 4194304)" error when I try to start it up.  To work around that I worked on getting the visibility recorder managers working.  At a high level the managers act like `Lwa352RecorderControl` but the main advantage is that they expose the recorders as a single entity for etcd.  So starting a recording now looks like:
```
from mnc.mcs import Client
c = Client()
// Start slow visibilities with 'drvs'
c.send_command('drvs', 'start', start_mjd='now', start_mpm='now')
// Stop fast visibilities with 'drvf'
c.send_command('drvf', 'stop', stop_mjd='now', stop_mpm='now')
```
There are also "summary" and "info" monitoring points for the managers that give overall visibility recording status:
```
>>> from mnc.mcs import Client
>>> c = Client()
>>> c.read_monitor_point('summary', 'drvs')
<MonitorPoint timestamp='2022-07-27 00:24:07.773262', value='error',
    unit=''>
>>> c.read_monitor_point('info', 'drvs')
<MonitorPoint timestamp='2022-07-27 00:24:07.773262', value='drvs7601:
    error (Only 9 of 10 threads active); drvs7602: normal (System
    operating normally); drvs7701: normal (System operating normally);
    drvs7702: normal (); drvs7801: normal (); drvs7802: normal (System
    operating normally); drvs7901: normal (System operating normally);
    drvs7902: normal (System operating normally); drvs8001: normal
    (System operating normally); drvs8002: normal (); drvs8101: normal
    (System operating normally); drvs8102: normal (System operating
    normally); drvs8201: shutdown - stale (System has been shutdown);
    drvs8202: shutdown - stale (System has been shutdown); drvs8301:
    shutdown - stale (System has been shutdown); drvs8302: shutdown -
    stale (System has been shutdown)', unit=''>
```

### Beams
Power beams seem to work but I haven't tested them extensively.  The version of the `xengine_beamformer_control` module in the calim branch has all of the updated IP/port values needed for calim.

### Outstanding Issues
 * The big one right now is the packet loss.
 * I'm not completely happy with the manager's "summary" and "info" fields and how they present the status.
 * `xengine_beamformer_control` is out of sync with `mnc_python` again and even with itself depending on whether you record to calim or gpu09.

caseyjlaw, 2022-07-28:
Thanks for documenting this well, Jayce.

jaycedowell, 2022-07-28:
* The packet loss has been fixed - I needed to update /etc/sysctl.conf to increase the network buffer sizes.  Sort of related: I also made a small change to /etc/systemd/logind.conf to set "RemoveIPC=no" since not doing that sometimes causes problems with Bifrost.
* I'm happier with the manager's summary and info now.  Info in particular is much more focused on what the actual problems are, e.g.:
 ```
>>> c.read_monitor_point('info', 'drvs')
<MonitorPoint timestamp='2022-07-28 00:55:52.044020', value='1 sub-
    bands error: drvs8302=Packet loss during receive >10% (50.0%
    missing)', unit=''>
 ```
* I addressed a problem that Marin was running into with the manager not being able to start or stop recordings.  That was related to the fast manager try to act like a second slow manager due to a missing `--quick` in the service file.
* I also addressed a problem stopping recordings at the individual recorder pipeline level that was related to the scheduling granularity being a poor match to the integration time for the slow visibilities.

That should be everything that I know of except for `xengine_beamformer_control`.

dgary50, 2022-08-04:
I had trouble today (and yesterday) with
    p = Lwa352CorrelatorControl(['lxdlwagpu0%d' % i for i in range(1,9)], npipeline_per_host=4)
on pipeline@lwacalim01.  It gave a connection failed, so I restarted the daemon on lxdlwagpu01 but it still gave the same failure.  I also modified the start of the range to 2 and then 3 and all give the same connection error, so maybe something else is going wrong than just the daemon?

jaycedowell, 2022-08-04:
I'm seeing the same thing when I tried it just now.  What's interesting is the error message:
```
2022-08-04 16:44:28 [ERROR   ] host lxdlwagpu01 (pipeline None) failed to respond
```
@jack-h why would pipeline be "None"?

jack-h, 2022-08-04:
Pipeline is "none" for commands sent to the main xengine controller (the service that start a pipelines).

I think on this ocassion you just need to update the control library on the calim server(s) https://github.com/realtimeradio/caltech-bifrost-dsp/tree/main/pipeline-control

jaycedowell, 2022-08-04:
Yep, that did the trick.

jaycedowell, 2022-08-04:
A couple more updates:
 * I've updated the manager to add in `op-type` and `op-tag` monitoring points to help keep tags on the recorders.  `op-type` tells what is happening (recording vs idle) and `op-tag` tells where things are being recorded to.
 * The slow visibility quota has been increased to 2 TB.  That should be enough for about 7 hours of recording.
 * I've put in places the changes to the measurement sets that @dgary50 asked for.  They *have not* been tested.
 * I need to give some more thought to how the antenna mapping is handled for the fast visibility mode so that the recorders can keep up with any dynamic changes.

jaycedowell, 2022-08-06:
I ran a quick recording this afternoon and the structure of the measurement sets look ok.

rlbyrne, 2022-08-10:
We just increased the slow visibility quota to 8 TB to facilitate a 24 hour data capture.

caseyjlaw, 2022-09-22:
Updated code on lwacalim00 for simplified control. Only really tested for slow visibilities.

```
> from mnc import control
> con = control.Controller('config/lwa_config_calim.yaml')  # path to config in mnc-python repo
> con.stop_xengine()
> con.start_xengine()
> con.start_dr()
> con.stop_dr()
```

caseyjlaw, 2022-09-30:
This is largely working as expected. I'll update the relevant docs on SharePoint and close this issue.

closed 2022-09-30 by caseyjlaw

caseyjlaw, 2023-07-07:
We set ovro_data_recorder up in the lwa-shell repo for uniform deployment. I migrated the services over to run that version, so the repo on cal-im at ~pipeline/ovro_data_recorder should no longer be used.
One residual issue is that the recorders use the `datarecorder` environment, but code is deployed to `deployment`. What's preferred: move recorders into `deployment` or have software installed to `datarecorder`?

yupinghuang, 2023-07-07:
Data recorder uses bifrost which was built by hand under datarecorder. You
can move it but you might have some fun with patchelf