ovro-lwa/lwa-issues #136: Service management on Calim

rlbyrne, 2022-08-10:
New script functionality that would be helpful is:
- Script for daemon reload
- Script to start service on all nodes 01-08 (it looks like start_dr_services.sh does not use all nodes)

Also: Can we add read permission for the ubuntu user in `/home/pipeline/ovro_data_recorder/services/`?

jaycedowell, 2022-08-11:
* There is a restart_dr_services.sh but it operates with the caveat that it only restarts _running_ services.  If you want to restart _all possible_ data recorder services then a stop_dr_services.sh followed by a start_dr_services.sh is what you are looking for.
 * We could think about setting up something like pdsh to help run commands across multiple nodes in parallel with a single command.

For your final point maybe the thing to do is change the services such that they are user services that can be managed from the pipeline account.  I'll look into that.

jaycedowell, 2022-10-12:
start/stop/restart_dr_services.sh now takes has -a flag to run the operation on lwacalim01 through lwacalim08 with a single command.

closed 2022-10-21 by jaycedowell