ovro-lwa/lwa-issues #138: astm-oss2 does not start; SOL console not working

yupinghuang, 2022-08-24:


yupinghuang, 2022-08-25:
From Rick:
"oss2 is brain dead. No output via the terminal/keyboard. All disk lights are on. When pressing the power button, all disk lights instantly go off. When powering on, all disk lights go on instantly and remain on. No output on terminal. It's not obvious where the OS disks are. Maybe internally?"

"oss2:
  Nothing appears on the screen. Not even the initial Post so won't be able to get into the BIOS. This looks worse than a disk failure since the BIOS
  is dead or something is preventing the BIOS from starting."

----------------------------------------------
I tried flashing the BIOS today via IPMI and it did not change anything. Now the IPMI interface shows no sensor working and no MAC addresses (it previously did show the MAC address). Rick thinks it is a motherboard issue.

rh-codebase, 2022-08-31:
oss1 had to be shutdown and removed to get to oss2. oss2's CMOS battery voltage 3.1V so not the issue. removed oss2 from shelter and brought it back to bldg12. oss1 had 1 bad power supply so replaced it with a good one from oss2. powered oss1 back up. astm should be ready to use w/out oss2.

rh-codebase, 2022-09-01:
A new MB for oss2 was ordered last night and should arrive by this Fri.

rh-codebase, 2022-09-06:
New MB arrived today and installed. oss2 is online and reachable from astm13.

closed 2022-09-06 by rh-codebase