ovro-lwa/lwa-issues #140: Rick can't log into any calim nodes

yupinghuang, 2022-08-24:


yupinghuang, 2022-08-24:
I'll try powering them up through ipmi. Potential NIS troubles.

yupinghuang, 2022-08-24:
Rick powered up the head node and fixed .ssh permissions on calim02 and things seemed mostly fine.

calim02's 40GbE card did not seem to come back up.
```
(base) [ubuntu@lwacalim01 ~]$ lspci | grep -i ethernet
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
c1:00.0 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
c1:00.1 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
[ubuntu@lwacalim02 ~]$ lspci | grep -i ethernet
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
```

All other nodes did get their IP addresses.

yupinghuang, 2022-08-24:
Looks like a few calim nodes powered up between 11h and 13h and they connected to the NIS running on calimhead just fine. Nodes that came up around 13:26 (03, 04, 09, 05, 06, 07, 08) failed to connect to the NIS and the ypbind service did not start.

A `pdsh -w lwacalim[00-10] 'sudo systemctl restart ypbind'` seems to fix things. `pdsh -w lwacalim[00-10] 'ypcat passwd'` returns on all nodes.

rh-codebase, 2022-08-26:
closing this and creating a separate issue regarding the 40G card on calim02

closed 2022-08-26 by rh-codebase