ovro-lwa/lwa-issues #141: maas not giving ip addresses via dhcp properly

yupinghuang, 2022-08-24:


rh-codebase, 2022-08-26:
will try to look at this again. DHCP was reporting 'no free leases'. However, this morning the snaps did get IP assignments.

rh-codebase, 2022-09-01:
static ip's have been statically assigned to the gpu nodes and their IPMI interfaces. The mapping is:
lxdlwagpu0X.mcs.pvt -> 10.40.0.5X
lxdlwagpu0Xipmi.mcs.pvt -> 10.40.0.6X
but can be accessed vis DNS using the above FQDN.

The last computer to switch is lxdlwacr.  @kathrynplant update ticket when this can be done and assign back to me. The switch take about 10min.

kathrynplant, 2022-09-02:
Hi Rick, is 4:00 today a good time for this?

rh-codebase, 2022-09-02:
sure. starting now. will let you know.

rh-codebase, 2022-09-02:
done. FQDN: lxdlwacr.mcs.pvt  IP: 10.40.0.50    
         FQDN: lxdlwacripmi.mcs.pvt  IP: 10.40.0.60

if all looks good, please close issue.

kathrynplant, 2022-09-02:
Thanks.  Just to check I understand correctly, I should always log in now by doing ssh 10.40.0.50 from maas instead of using a host name?

closed 2022-09-03 by kathrynplant