ovro-lwa/lwa-issues #149: ARX board 28 missing 1-wire temperature map

rh-codebase, 2022-09-10:
The 1-wire temperature map needs to be programmed into the board.

ldaddario, 2022-09-22:
All boards had the temperature sensor map (it should not be called '1-wire'  here) stored in on-board non-volatile memory when delivered.  This problem seems to have occured becuase the proper initialization procedure was not followed in the lab when the board's software was updated post-delivery.

I can fix this remotely whenever I have access to the RS485 bus that controls the boards.  Right now access is broken because of an issue with the etcd server.

ldaddario, 2022-10-26:
Actually, three installed boards are missing their temperature sensor mapping information; they are SN18 (at address 28), SN36 (address 29), and SN22 (address 30).  I have fixed these remotely.  The maps are
SN18  F07  (sensors 0,1,2 -> channels 16,1,8)
SN36  0F7
SN22  70F.
Note that the mapping goes with the physical board (identified by serial number), not with the address.

closed 2022-10-26 by ldaddario

ldaddario, 2022-10-26:
I fixed this remotely.