ovro-lwa/lwa-issues #151: lwacfg service unable to download Sharepoint Antenna Config Spreadsheet

rh-codebase, 2022-09-10:
Was working on 2022-06-24 but failure could be due to proposed rollout of OAUTH2 for Sharepoint.

rh-codebase, 2022-09-15:
since other docs in sharepoint are being migrated to goggle docs how about migrating the Antenna Config Spreadsheet? The API for Google Docs is likely more stable. Thoughts?

caseyjlaw, 2022-09-16:
I haven't played with either API. But the 2FA change does make it tricky to automate downloads from cloud, in general.
The rate at which we update this doc is low, so while I like the automation, I personally am ok with doing it manually via the browser.

rh-codebase, 2022-12-19:
Browser it is for now.
 I tried another API via:
pip3 install Office365-REST-Python-Client

which returned the following error:

An error occurred while retrieving token from XML response: AADSTS50158: External security challenge not satisfied. User will be redirected to another page or authentication provider to satisfy additional authentication challenges.
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    ctx_auth.acquire_token_for_user(USER, PWD)   
  File "/home/ubuntu/.local/lib/python3.6/site-packages/office365/runtime/auth/authentication_context.py", line 82, in acquire_token_for_user
    return self._provider.ensure_authentication_cookie()
  File "/home/ubuntu/.local/lib/python3.6/site-packages/office365/runtime/auth/providers/saml_token_provider.py", line 84, in ensure_authentication_cookie
    self._cached_auth_cookies = self.get_authentication_cookie()
  File "/home/ubuntu/.local/lib/python3.6/site-packages/office365/runtime/auth/providers/saml_token_provider.py", line 99, in get_authentication_cookie
    token = self._acquire_service_token()
  File "/home/ubuntu/.local/lib/python3.6/site-packages/office365/runtime/auth/providers/saml_token_provider.py", line 176, in _acquire_service_token
    token = self._process_service_token_response(response)
  File "/home/ubuntu/.local/lib/python3.6/site-packages/office365/runtime/auth/providers/saml_token_provider.py", line 201, in _process_service_token_response
    raise ValueError(self.error)
ValueError: An error occurred while retrieving token from XML response: AADSTS50158: External security challenge not satisfied. User will be redirected to another page or authentication provider to satisfy additional authentication challenges.

Same error as trying with the sharepy API via: pip3 install sharepy

Propose not to use Sharepoint for such docs for DSA2k.

closed 2022-12-19 by rh-codebase

caseyjlaw, 2022-12-19:
That is valuable input to Francois, as we start DSA2k planning.

rh-codebase, 2022-12-19:
It's arguable whether all docs/configs have to be programmatically available but these types I feel should be.