ovro-lwa/lwa-issues #153: Cannot start all GPU pipelines

rlbyrne, 2022-09-14:
We attempted to restart the GPU pipelines using the `lwa352_pipeline_control` python script and following instructions in the [System Use Recipe](https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Commissioning/[System-Use-Recipe.docx](https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Commissioning/System-Use-Recipe.docx?d=wa7dc43fa71f8468f8d22b00ef8c82f2f&csf=1&web=1&e=nalOPi)?d=wa7dc43fa71f8468f8d22b00ef8c82f2f&csf=1&web=1&e=nalOPi). When that didn't work, we tried restarting the control service and X-engine etcd clients following the instructions in the [Observer on Duty procedures document](https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Commissioning/[LWA-352_OoD_Procedures.docx](https://caltech.sharepoint.com/sites/ovro/projects/LWA%20Documents/Commissioning/LWA-352_OoD_Procedures.docx?d=wcefacc27d2564672962e8a11499a84f6&csf=1&web=1&e=52uQ3s)?d=wcefacc27d2564672962e8a11499a84f6&csf=1&web=1&e=52uQ3s). However, we were not able to start all GPU servers. Running the `p.pipelines_are_up(verbose=True)` command shows that only about half the pipelines are operating.

jaycedowell, 2022-09-15:
What errors were you getting when you tried this?  Just now I gave creating a `Lwa352CorrelatorControl` instance a shot and I seem to get either:
```
etcd3.exceptions.ConnectionFailedError: etcd connection failed
```
or a 
```
RuntimeError: Connection failed. Consider restarting lwa-xeng-etcd.service daemon on host lxdlwagpu##
```

For the `etcd3.exceptions` one I also see a:
```
    raise _InactiveRpcError(state)
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "etcdserver: request timed out"
        debug_error_string = "{"created":"@1663260663.849996389","description":"Error received from peer ipv4:10.40.0.253:2379","file":"src/core/lib/surface/call.cc","file_line":1074,"grpc_message":"etcdserver: request timed out","grpc_status":14}"
```
before that exception is thrown.  I'm not sure what this all means right now but I'm inclined to say it is an etcd problem.

rlbyrne, 2022-09-15:
That's strange, I wasn't getting any errors yesterday. The command seemed to run fine, but then the `p.pipelines_are_up(verbose=True)` output showed that many of the pipelines weren't running.

jaycedowell, 2022-09-15:
Was there any pattern to which nodes/pipelines were not running?

jack-h, 2022-09-15:
It looks like the `lwa-xeng-etcd.service` has a flaky connection to etcd. I just ran `systemctl status lwa-xeng-etcd.service` and serveral nodes were showing connection failures. This sometimes happens after power cycles if the clients try to connect to the etcd server before it is ready, but even restarting the services they keep dropping connections.
Sometimes I'm also seeing errors from the etcd client. The `pipelines_are_up` also talks through etcd, so your issue could have the same root cause (I restarted `bifrost_etcd_bridge.service` in case that was where the problem was). However, right now it seems that the etcd connection is just flaky for some reason. @rh-codebase any ideas?

rh-codebase, 2022-09-15:
I'm also seeing lots of stale temperature readings which stem from the service unable to get a response from etcd. Looking into it.

rlbyrne, 2022-09-15:
@jaycedowell I didn't notice any pattern

rh-codebase, 2022-09-16:
The stale temp. readings was a different issue. I wrote some scripts on lxdlwagpu09 to facilitate testing(i.e. restartBifrostService, restartLWA_xengservice, testLWApipeline etc). There are also a lot of warn message type being reported by etcd:

`Sep 16 16:23:46 etcdv3service startEtcdServer[841]: {"level":"warn","ts":"2022-09-16T16:23:46.562Z","caller":"v3rpc/interceptor.go:197","msg":"request stats","start time":"2022-09-16T16:23:46.167Z","time spent":"394.33047ms","remote":"10.40.0.52:59236","response type":"/etcdserverpb.KV/Put","request count":1,"request size":540,"response count":0,"response size":29,"request content":"key:\"/mon/corr/x/lxdlwagpu02/pipeline/0/udp_transmit/10\" value_size:485 "}
Sep 16 16:23:47 etcdv3service startEtcdServer[841]: {"level":"warn","ts":"2022-09-16T16:23:47.167Z","caller":"etcdserver/util.go:166","msg":"apply request took too long","took":"197.026773ms","expected-duration":"100ms","prefix":"","request":"header:<ID:7587865288246701906 > put:<key:\"/mon/corr/x/lxdlwagpu02/pipeline/0/udp_transmit/3\" value_size:483 >","response":"size:5"}
Sep 16 16:23:47 etcdv3service startEtcdServer[841]: {"level":"warn","ts":"2022-09-16T16:23:47.167Z","caller":"v3rpc/interceptor.go:197","msg":"request stats","start time":"2022-09-16T16:23:46.778Z","time spent":"388.51982ms","remote":"10.40.0.52:59236","response type":"/etcdserverpb.KV/Put","request count":1,"request size":537,"response count":0,"response size":29,"request content":"key:\"/mon/corr/x/lxdlwagpu02/pipeline/0/udp_transmit/3\" value_size:483 "}
`

Has any of the monitor info increased?  etcd will issue these warning whenever the time to execute is > 100ms.  The file I/O requirements(specifically fdatasync) look to be reasonable. I ran the recommended test as described [here](https://www.ibm.com/cloud/blog/using-fio-to-tell-whether-your-storage-is-fast-enough-for-etcd). With all containers on MCS down, the 99th percentile time for fdatasync = 5.58ms which is below the 10ms recommendation. In contrast, my laptop reported 5.6ms and the DSA110 system will all containers running including etcd reported: ~ 33ms.  etcd on DSA110 rarely(1 or 2 every few minutes) reports any exec times > 100ms.  Since the LWA fdatasync times reported by fio look good I don't suspect a disk issue.

Can the monitor data spewed be reduced to the bare minimum and the test scripts run to see if it makes a difference? Only etcd is up at the moment(i.e. All other services related to MCS have been stopped). The test script is currently config. to start up 2 pipelines and it is consistently failing so also take a look at the script for sanity.

caseyjlaw, 2022-09-16:
I used dsa_store to look some x-engine monitor points. I could see services updating those values. Anecdotally, it did seem to get quite slow for some queries.
I've stopped the `lwa-xeng-etcd` services on lxdlwagpuxx. Do you want to try a test on throughput now?

jack-h, 2022-09-16:
I don't know what RFI stuff is running but that could potentially be quite
heavy depending how it is being used.

caseyjlaw, 2022-09-16:
I don't believe any RFI reader code is running. Can you confirm @greghell ?

greghell, 2022-09-16:
That's right, nothing is running at the moment.

rh-codebase, 2022-09-16:
I forgot to include the output from running the testLWApipeline script:

`(casa) ubuntu@lxdlwagpu09:~$ ./testLWApipeline
Creating 2 LWA352CorrelatorControl objects
Stopping pipelines
Waiting 10 sec to Start pipelines
Starting pipelines
2022-09-16 16:28:03 [INFO    ] Pipelines all appear to be ready after 29.0 seconds
Checking pipelines in 5 sec
Checking...
lxdlwagpu01:0 - up? True
lxdlwagpu01:1 - up? True
lxdlwagpu01:2 - up? True
lxdlwagpu01:3 - up? False
lxdlwagpu02:0 - up? False
lxdlwagpu02:1 - up? False
lxdlwagpu02:2 - up? False
lxdlwagpu02:3 - up? False
(casa) ubuntu@lxdlwagpu09:~$ 
`

So, no exceptions are bubbling up but I'm not sure if there is any logging somewhere explaining why the pipelines didn't start. The INFO mesg seems to say the pipelines appear to be up. What is being used to determine this?

caseyjlaw, 2022-09-16:
The correlator pipeline control will look at the time stamp in etcd related to the bifrost service status. If any are stale/late, then the `pipelines_are_up` is False.
FWIW, we are seeing intermittent failures for f- and x-engine tests that use etcd. It is very consistent with your suggestion that etcd latency or intermittency is the problem. Is there a way to actually see how much use etcd is getting and confirm that it is higher than expected (e.g., from DSA-110 experience)?

rh-codebase, 2022-09-16:
possibly through the etcd metrics. It needs plumbing but it could show something.
So is the monitor vol. from these services the same before and after the last major power shutdown? If so, then there is likely some other issue going on. Before the power shutdown, when this was working, we had a much more loaded system( webui, shutdown, etcd2db etc). Now with only 2 bifrost and 2 xeng services, the 2 pipelines fail. 

What logic is being used to produce the INFO message above? There seems to be a 60sec timeout associated with it as I've seen messages stating 60sec have elapsed ....

caseyjlaw, 2022-09-16:
Traffic is definitely down, so I totally agree something in etcd is not right.
I didn't write that test code, so I'm not sure. I'm guessing it is some Python logging.

rh-codebase, 2022-09-17:
got etcd metrics up in grafana but still couldn't reliably start pipelines. Stopping the data recorders seemed to help but repeated tries with the pipelines never quite took. At times just starting 2 or 3 would work but never got 4 up. The WAL fsync still looks excessively high though there was a time(~ 15:35 PDT) where it dropped below 20ms and I could get 2 pipelines 2-3 times in a row. Then the WAL fsync times started to slowly rise and no luck afterwards. Done for the day.

jack-h, 2022-09-17:
My interpretation of this message chain is that the issue is with the etcd
server(?)

What (if anything) would you like me to do to work around this?
The quick/ dirty thing is to temporarily move F/X control to a different
etcd server (I could spin one up on gpu09 or the cr node).
Longer term I could make a mechanism to support xengine comms which doesn't
rely on etcd, in much the same way as the snaps can be controlled without
etcd if necessary.

caseyjlaw, 2022-09-17:
Running the etcd server on different hardware is a good idea. I gather you've done this plenty while developing your F/X control development.
Feel free to give this a try.

jack-h, 2022-09-18:
1. I've installed (using `apt`) and started an etcd server (version `3.2.17`) on the cosmic ray node `lxdlwacr.sas.pvt`
2. On The cosmic ray node, and gpu nodes 1 through 8 (not including 9), I have redirected `etcv3service.sas.pvt` to `10.40.0.50` (which is `lxdlwacr`). This probably isn't the right thing to do, but it meant i didn't have to change any config files anywhere.
3. I then restarted (on the GPU nodes) `lwa-xeng-etcd.service`, and `bifrost_etcd_bridge.service`. The former processes correlator commands. The latter makes pipeline status keys available to redis. I restarted (on the CR node) `lwa-feng-etcd@*.service` (one service per SNAP board) which facilitate etcd comms to the SNAPs.
4. From `lxdlwacr` I can instantiate and use an etcd Fengine control object, and the 4 correlator pipelines on all 8 nodes start without issue.

Hopefully this is enough to get things on the air. Note that from any machine other than the CR node (eg. `gpu09`, or any of the calim systems) you'll need to point the etcd host argument when instantiating F/X control objects to `lxdlwacr.sas.pvt`. Note also that my hack of the hosts file on the CR node and gpu1-8 means that these nodes will no longer be able to reach the "real" (original) etcd server by its `etcdv3service` hostname.

nmahesh1412, 2022-09-19:
Just to add, when @rlbyrne and I tried creating a Lwa352CorrelatorControl instance, I got the following:
 Lwa352CorrelatorControl('lxdlwagpu09', npipeline_per_host=4)
2022-09-16 22:52:29 [ERROR   ] host l (pipeline None) failed to respond to etcd command!
2022-09-16 22:52:29 [ERROR   ] Control interface on host l failed to respond to ping!
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-2-8f093773b7be> in <module>()
----> 1 p = Lwa352CorrelatorControl('lxdlwagpu09', npipeline_per_host=4)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in __init__(self, hosts, npipeline_per_host, etcdhost, log)
     79                                                             pipeline_id=pipeline_id,
     80                                                             etcdhost=etcdhost,
---> 81                                                             log=log,
     82                                                         )]
     83         self.npipeline = len(self.pipelines)

~/anaconda3/envs/casa/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in __init__(self, host, pipeline_id, etcdhost, log)
    345         self.beamform_vlbi_output = BeamformVlbiOutputControl(*args)
    346         if not self.check_connection():
--> 347             raise RuntimeError("Connection failed. Consider restarting lwa-xeng-etcd.service daemon on host %s" % host)
    348 
    349     def start_pipeline(self):

RuntimeError: Connection failed. Consider restarting lwa-xeng-etcd.service daemon on host l

nmahesh1412, 2022-09-19:
So just to add to @jaycedowell question - It was the latter error (Runtime error) of the two errors you got

caseyjlaw, 2022-09-19:
Thank you, @jack-h, for getting that set up.
Nivedita, Ruby, and I tried to get everything else set up with the new etcd server on lxdlwacr. We could see that the x-engines pipelines were running for the new etcd server location. However, we had trouble getting the data recorders running with the new server location.
@jaycedowell We updated the `mnc.common` code to point to a new etcd server on lxdlwagpu09. After that update, we restarted the dr-vslow recorders and ran the ezdr code to create a data recorder client. However, the auto-discovery of data recorders failed. Did you deploy the changes related to the new scheduler scheme? If so, then ezdr would fail, right?

@rh-codebase Have you had any luck testing the performance of etcd at etcdv3service? Ideally, we'd fall back to that.

jaycedowell, 2022-09-19:
The new visibility managers shouldn't interfere with the old ezdr auto-discovery/scheduling.

caseyjlaw, 2022-09-19:
@jaycedowell Ok, thanks. If I update the etcd server name in `mnc.common`, should that propagate to all data recorder code and services?

jaycedowell, 2022-09-19:
Yes, it should but there is a lot of weirdness with the pipelines on lxdlwagpu09.  The normal log looks ok but if I `journalctl -u dr-vslow-1 --no-pager` I get:
```
Sep 19 14:46:43 lxdlwagpu09 systemd[1]: Started OVRO-LWA Slow Visibilties Recorder - Band 1.
Sep 19 14:46:44 lxdlwagpu09 dr-vslow-1[32311]: Read antpos from xlsx file in repo
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: Exception in thread Thread-8:
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: Traceback (most recent call last):
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 916, in _bootstrap_inner
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:     self.run()
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 864, in run
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:     self._target(*self._args, **self._kwargs)
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/ovro_data_recorder/monitoring.py", line 554, in main
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:     self.status.main(once=True)
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/ovro_data_recorder/monitoring.py", line 405, in main
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]:     dfree = 1.0*free.value / total.value
Sep 19 14:46:46 lxdlwagpu09 dr-vslow-1[32311]: AttributeError: 'NoneType' object has no attribute 'value'
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: ++++++++++++++++ seq0     = 165847932
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:                  time_tag = 326070303452037120
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: ******** CFREQ: 15312500.0
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: 2022-09-19 21:46:49        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double)         Requested JD 59841.9 is outside the range of the IERS (Earth axis data) table.
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: 2022-09-19 21:46:49        INFO        MeasIERS::fillMeas(MeasIERS::Files, Double) +        Calculations will proceed with less precision
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: Exception in thread Thread-5:
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: Traceback (most recent call last):
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 916, in _bootstrap_inner
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     self.run()
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/threading.py", line 864, in run
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     self._target(*self._args, **self._kwargs)
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/ovro_data_recorder/dr_visibilities.py", line 464, in main
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     uvw = get_zenith_uvw(self.station, LWATime(time_tag, format='timetag'))
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/ovro_data_recorder/lwams.py", line 66, in get_zenith_uvw
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     position = dm.position('ITRF', *['%.6fm' % v for v in ant.ecef])
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/casacore/measures/__init__.py", line 189, in position
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     loc['m0'] = dq.quantity(v0)
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:   File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/casacore/quanta/quantity.py", line 57, in quantity
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]:     return Quantity(from_string(args[0]))
Sep 19 14:46:48 lxdlwagpu09 dr-vslow-1[32311]: RuntimeError: in QuantumHolder::fromString with input string "nanm": Illegal input units or format
Sep 19 14:47:08 lxdlwagpu09 dr-vslow-1[32311]: /home/ubuntu/ovro_data_recorder/dr_visibilities.py:336: RuntimeWarning: divide by zero encountered in log10
Sep 19 14:47:08 lxdlwagpu09 dr-vslow-1[32311]:   im = self._plot_spectra(time_tag, freq, 10*numpy.log10(adata))
```
"ERROR2" usually comes from a problem reading a key from etcd and that appears to be causing some of the other error messages.  The failure to generate the _(u,v,w)_ coordinates is really strange.  It could be related to lwa_antpos falling back to the spreadsheet?

jaycedowell, 2022-09-19:
After digging into this a little I finally was able to get an error out of etcd:
```
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.RESOURCE_EXHAUSTED
	details = "etcdserver: mvcc: database space exceeded"
	debug_error_string = "{"created":"@1663625888.628152084","description":"Error received from peer ipv4:10.40.0.50:2379","file":"src/core/lib/surface/call.cc","file_line":1074,"grpc_message":"etcdserver: mvcc: database space exceeded","grpc_status":8}"
```

jack-h, 2022-09-20:
@jaycedowell I'm not sure what you did to try and grow the database (I tried some stuff too). But, ultimately, nothing worked until I ran (on `lxdlwacr`):

```
apt install etcd-client
export ETCDCTL_API=3
etcdctl alarm disarm
```
So it seems very possible that you did fix it, but needed to reset some alarm first(?) I was led here by https://github.com/etcd-io/etcd/issues/11947

jack-h, 2022-09-20:
For the record, `/lib/systemd/system/etcd.service` on `lxdlwacr` now includes the following lines

```
LimitNOFILE=524288
ETCD_QUOTA_BACKEND_BYTES=17179869184
MemorySoftLimit=17000000000
MemoryLimit=17000000000
```
Almost certainly some of them are not required or are unnecessarily high.

caseyjlaw, 2022-09-20:
We got closer today, but not quite there yet.

@jaycedowell I have found the origin of the uvw calculation error. The antenna positions included some bad values from a recent update. Fixed now.
I am still seeing some of these JSON errors:
```Sep 20 10:56:02 lxdlwagpu09 dr-vslow-1[14211]: ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'```

I can reproduce that error when using the Client class you created to look at monitor points:
```
> from mnc.mcs import Client
> c = Client()
> c.list_monitor_points('drvs')
ERROR2: the JSON object must be str, bytes or bytearray, not 'NoneType'
```

Does that trigger any ideas from you?

caseyjlaw, 2022-09-20:
@jack-h After our call, I started/stopped data recorders and pipelines. Now it is reporting the `database space exceeded` error again.
I tried to reset it as described above, but to no effect.

caseyjlaw, 2022-09-20:
@rh-codebase We are struggling with setting up a parallel etcd server. Any idea on whether the original server is performing as expected? We could try moving back to that service, if you think it worthwhile.

rlbyrne, 2022-09-20:
@caseyjlaw Is the UVW calculation error new since the shutdown? Should I be worried that UVWs in the 24 hour run I took 5 weeks ago are bad?

caseyjlaw, 2022-09-20:
@rlbyrne Sorry, I didn't describe it well. No error in values. An error prevented the data recorder code from running while trying to calc the uvw.

rh-codebase, 2022-10-03:
the etcd database can be controlled by config etcd to have depth 1. That is, not to store historical values. I'll look up the config option. @jack-h I think moving etcd to cr shows it's not etcd but some other issue on the mcs host which is preventing etcd from performing writes within its 100ms spec.  An etcd cluster may provide robustness but there's a performance price to pay for it.

rh-codebase, 2022-10-04:
here are options to use on etcd to eliminate database growing:

--auto-compaction-retention 1 --auto-compaction-mode revision

jack-h, 2022-10-11:
I think this makes sense - there are no xengine pipelines on gpu09, and
therefore isn't a control service running which you can connect to.

rh-codebase, 2023-06-09:
closing out until #92 is closed.

closed 2023-06-09 by rh-codebase