ovro-lwa/lwa-issues #156: DNS needs config on calim?

yupinghuang, 2022-09-16:
This happens after the IP reconfig. calim can't resolve DNS but does have access to the internet.
```
(base) [ubuntu@lwacalimhead ~]$ ping www.cloudflare.com
ping: www.cloudflare.com: Name or service not known
(base) [ubuntu@lwacalimhead ~]$ ping 104.16.124.96
PING 104.16.124.96 (104.16.124.96) 56(84) bytes of data.
64 bytes from 104.16.124.96: icmp_seq=1 ttl=52 time=32.3 ms
64 bytes from 104.16.124.96: icmp_seq=2 ttl=52 time=32.2 ms
^C
--- 104.16.124.96 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 32.181/32.245/32.310/0.190 ms
```

for comparison on lwamaas
```
ubuntu@lwamaas:~$ ping www.cloudflare.com
PING www.cloudflare.com (104.16.124.96) 56(84) bytes of data.
64 bytes from 104.16.124.96 (104.16.124.96): icmp_seq=1 ttl=53 time=31.8 ms
64 bytes from 104.16.124.96 (104.16.124.96): icmp_seq=2 ttl=53 time=31.9 ms
^C
--- www.cloudflare.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 31.897/31.947/31.998/0.185 ms

```

yupinghuang, 2022-09-20:
Added the following line to `/etc/resolv.conf` on all calim nodes. Now the nodes can resolve domain names. Reassigning to Rick to make sure that this is the right thing to do.

```
nameserver 192.100.16.2
```

caseyjlaw, 2022-09-21:
@yupinghuang Should Cal-Im be available after this change? We were hoping to test data recording there, now that it is working on gpu09.

rh-codebase, 2022-10-03:
Using nameserver 192.100.16.2 on calim nodes will on ly allow them to resolve OVRO and outside hosts. 192.100.16.2 has no knowledge of hosts within the LWA network controlled by MAAS. If you need both, then the nameserver on10.40.0.4 should provide it.

I'm able to ssh to lwacalim01 vial lwamaas as user ubuntu.

yupinghuang, 2022-10-04:
On all calim nodes now it looks like
```
$ cat /etc/resolv.conf 
# Generated by NetworkManager
nameserver 10.40.0.4
search sas.pvt mcs.pvt ant.pvt
```
Closing.

closed 2022-10-04 by yupinghuang