ovro-lwa/lwa-issues #159: Missing 40G interface on lwacalim06

jaycedowell, 2022-09-29:
It's not there:
```
(base) [pipeline@lwacalim06 ~]$ lspci | grep -i ether
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
```

rh-codebase, 2022-10-03:
similar to issue #144. Manual removal and inserting the card will likely be needed.

rh-codebase, 2022-10-07:
logging in and rebooting system brought back interface:

(base) [ubuntu@lwacalim06 ~]$ lspci | grep -i ether
44:00.0 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
44:00.1 Ethernet controller: Intel Corporation Ethernet Controller X710 for 10GBASE-T (rev 02)
c1:00.0 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)
c1:00.1 Ethernet controller: Intel Corporation Ethernet Controller XL710 for 40GbE QSFP+ (rev 02)

caseyjlaw, 2022-11-03:
lwacalim06 is currently down. Maybe this is a broader issue?

yupinghuang, 2022-11-03:
on it

yupinghuang, 2022-11-03:
IPMI reports that the host was powered off so I turned it on. Seems like lwacalim06 has been off since the hour after Oct 27 15:10 UTC. Nobody was logged in at the time. These issues appear unrelated.

caseyjlaw, 2022-11-03:
Thank you for getting that back up.

closed 2022-11-03 by caseyjlaw