ovro-lwa/lwa-issues #160: Issues with Beamforming observations and Track Command

chhabrasherry, 2022-10-12:
We followed the system use recipe to start the x engine. Issued the track command and started the new terminal to issue the recording command:
1. The log showed 50.7% right after starting the x_engine. This loss was reported to be 45.5 % last week (04/10/2022). Additionally, the log shows a “queue size: 1” before we tried to start recording the data. The same queue also appeared in the previous Ood's log. We did not know how to check when the previous recording command was queued and how to abort it.
** a screenshot of the log is attached**
2. The second terminal requires rerunning the con.start_xengine_bf(track=True) command before recording can be started. This is an additional step that was not required on gpu09.
3. We see 2 logs for the same date, namely: dr-beam-1.lwacalim01.log.2022-10-09 and dr-beam-2.lwacalim01.log.2022-10-09. We did not submit any commands that should start beam 2.

**Issues with the Track command:**
1. The track command requests a target, even though the default=None is supposed to go to ‘zenith’ automatically. We tried rerunning it with target = ‘zenith’, but that was not actually a recognized target to track.
2. The documentation is inaccurate regarding the target’s coordinates. Is it RA, DEC or Alt, Az? If RA DEC, what units are accepted for RA?
[image]

jaycedowell, 2022-10-12:
> The log showed 50.7% right after starting the x_engine. This loss was reported to be 45.5 % last week (04/10/2022). Additionally, the log shows a “queue size: 1” before we tried to start recording [...]

I think the lingering "queue size: 1" is a DR problem where old recording commands aren't removed from the queue.  I'll look into it.

> We see 2 logs for the same date, namely: dr-beam-1.lwacalim01.log.2022-10-09 and dr-beam-2.lwacalim01.log.2022-10-09. We did not submit any commands that should start beam 2.

The log files are generated as long as the DR services are running.  They get populated regardless of whether or not a beam is used.

> The track command requests a target, even though the default=None is supposed to go to ‘zenith’ automatically. We tried rerunning it with target = ‘zenith’, but that was not actually a recognized target to track.

I looked at [mn.control.Controller.start_xengine_bf](https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L192) and, yes, it will not track zenith.  I guess that makes sense since there is nothing to track at zenith because it is a fixed point.  @caseyjlaw  the documentation should probably be updated to reflect that you need a real target to track in order to track.

> The documentation is inaccurate regarding the target’s coordinates. Is it RA, DEC or Alt, Az? If RA DEC, what units are accepted for RA?

Where is this discrepancy at?

rh-codebase, 2022-10-12:
Maybe 'track' is the wrong command to point a beam at a fixed Az,El but this seems like a worth while requirement. How else can one verify beamwidth/beam shape?

jaycedowell, 2022-10-12:
Support for alt,az is available through the lower level `xengine_beamformer_control` module but not currently through `start_xengine_bf`.

caseyjlaw, 2022-10-14:
I've updated the system use recipe and code to clear this up. New syntax is to use the `track` argument to define how the `target` argument is interpreted. If not tracking, target is (az, el), otherwise, it is (ra, dec) or target name.

iveydavis, 2022-10-14:
> 

Just tried implementing this (tracking the Sun and just pointing at zenith) but something else seems to be broken
[image]
[image]
This looks like it maybe an unrelated problem-- we were transitioning back to the original etcd server I think. Maybe that could be responsible? I can submit a separate issue related to that since something seems to have affected slow visibilities recording as well.

rh-codebase, 2022-10-14:
Yes. Please submit the separate issue.

jaycedowell, 2022-10-21:
@caseyjlaw and I did some work on this. I would like to think that this is resolved.

closed 2022-10-21 by caseyjlaw