ovro-lwa/lwa-issues #162: Xengine Control + Data Recording Issues

iveydavis, 2022-10-14:
I ran into a few issues/peculiarities while running observing tests following the system use recipe.

note, `con` is defined from:
```
from mnc import control
con = control.Controller('config/lwa_config_calim.yaml')
```
First, it's not clear that `con.stop_xengine()` actually stopped the x-engine. This is the output from checking the status after running it:
[image]

Which appears to be the same as the output when running `con.start_xengine()`  (first screenshot is the immediate output of running this, the second screenshot is the result of checking the status):

[image]
[image]

I tried running `con.start_dr(recorders=['drvs'])` and got the following message:
[image]

It didn't record anything and it's not clear to me why I would get a message about the beamformer when I'm just trying to record a slow visibility.  I tried to 'circumvent' the problem by getting the beamformer ready to record too, but got the following error:
[image]

Assigning a target to track returns the same error:
[image]

After all the failures, I tried running `con.stop_xengine()` again, but it's still unclear to me that it actually stopped the xengine. I'll need to look at the logs more closely.

jaycedowell, 2022-10-18:
I just ran into something similar to your `con.start_xengine_bf()` error but through a direct interaction with the `mnc.xengine_beamformer_control` module.  For me the root cause of the `KeyError: 'chan0'` came from not all of the xengine pipelines receiving data.  No data flowing into the pipeline, no way to know what `chan0` is.  This happened because I naively used the old `lwa_corr_config_haarp.yaml` configuration for the SNAPs which was pre-the IP addresses getting changed on lxdlwagpu##.  Wrong IP address, no data into the pipelines.

iveydavis, 2022-10-18:
Are you running this on gpu09 or calim00? I'd hoped to use `mnc.xengine_beamformer_control` to investigate the problem, but I can't check the status of the data recording services on calim00.

jaycedowell, 2022-10-18:
I'm on calim00 right now.  Yeah, you cannot directly check on the data recorders  a la `like_bmon.py` from 00.  The monitoring points should work through a `mnc.mcs.Client` instance though.

jaycedowell, 2022-10-21:
@caseyjlaw and I did some work on this. I would like to think that this is resolved.

caseyjlaw, 2022-10-21:
Yes, this should be resolved by the fix to the configuration file change (on cjl/dev). I'll merge that in to main.

closed 2022-10-21 by caseyjlaw

caseyjlaw, 2022-11-03:
@jaycedowell I think we ran into this issue again after reprogramming snaps. The `mnc-python` config file didn't change, but it now reports this error when running beamformers and a sync error when configuring for slow visibilities.
I'll test again, but can you recommend a centralized way to check whether the config is mapping f- to x-engine correctly?

caseyjlaw, 2022-11-03:
I fixed a configuration error and rerun an observing test with slow vis and power beams 1 and 2. No issues.
The error must have been copied in when resolving this issue (snap02 was not included and a non-existent "snap12" was added).

caseyjlaw, 2022-12-14:
I saw this error again and noticed in the dashboard that one of the pipelines on gpu3 was down. Stopping and reconfiguring the xengines cleared the issue.