ovro-lwa/lwa-issues #167: Verify f-eng zero/noise behavior

caseyjlaw, 2022-10-25:
Set a subset of inputs to zero, and the others to a noise generator.
Verify that the bitstats and autocorrelations of only the expected channels are zero.

rlbyrne, 2022-10-26:
Done, with `./test_internal_noise.py --zero snap03`
Code is available [here](https://github.com/realtimeradio/caltech-lwa/blob/66d771fe4e91c4760df37996d7c4d431a7f135b2/control_sw/tests/test_internal_noise.py)

rlbyrne, 2022-11-01:
I just noticed that this test initially fails after reprogramming with the following:

> 		from lwa_f import snap2_fengine
> 		f = snap2_fengine.Snap2Fengine('snap03')
> 		f.program()
> 		f.initialize(read_only=False)`

rlbyrne, 2022-11-02:
Fixed by Jack here: https://github.com/realtimeradio/caltech-lwa/commit/9da8c4ae018f821769fbdec9686f1955c330c3b6