ovro-lwa/lwa-issues #169: f-eng EQ Coefficient verification

caseyjlaw, 2022-10-25:
(using inbuilt white-noise ADC test-vectors) 

- Zero all correlator inputs; verify correlator output is zero 
- Zero subset of polarizations; verify expected correlator outputs are zero 
- Zero subset of frequency channels; verify expected correlator outputs are zero 
- Set EQ coefficients for a target RMS. Verify the correlator outputs are expected levels

rlbyrne, 2022-10-28:
- [x] Zero all correlator inputs; verify correlator output is zero
- [ ] Zero subset of polarizations; verify expected correlator outputs are zero
- [ ] Zero subset of frequency channels; verify expected correlator outputs are zero
- [ ] Set EQ coefficients for a target RMS. Verify the correlator outputs are expected levels

rlbyrne, 2022-11-02:
It may be worthwhile to look at the zero polarization test downstream to confirm that the polarization indexing is correct.

caseyjlaw, 2023-01-18:
These tests used correlator module in f-engine.

caseyjlaw, 2023-02-09:
Partial verification of f-eng control showing up in recorded visibilities...
I turned off ant-pols in the f-engine and looked at written MS files. This plot shows a subset of antennas (pol a top, pol b bottom) with color showing a sequence in time. Indeed, pol a is being turned off (pol b already off in this plot).

[image]

caseyjlaw, 2023-08-30:
This has been exercised well by Larry's script, too.