ovro-lwa/lwa-issues #170: Verify f-eng channel frequency labelling

caseyjlaw, 2022-10-25:
Apply sinusoid. Verify it lands in expected channel in on-board SNAP spectra

caseyjlaw, 2022-10-28:
No feature in f-eng for sinusoid injection yet. Would require signal generator in lab.
Could also use RFI as frequency reference.

caseyjlaw, 2023-02-09:
@greghell Could you help me with identification of known RFI?

I can make an auto-corr spectrum with the f-engine and with recorded (MS) data. It looks like this:
[image]

The peaks line up, which is good. Now I just want to verify that the frequency values in the f-eng autocorrs map correctly to known RFI.

I have a list of frequencies for the peaks. Can you confirm that some of these are correct for known interferers?

[15.4440918 , 15.49194336, 15.77905273, 15.97045898, 17.11889648, 17.50170898, 18.14770508, 18.91333008, 20.01391602, 21.04272461, 21.09057617, 21.49731445, 21.80834961, 22.4543457 , 22.95678711, 23.74633789, 23.77026367, 24.91870117, 24.94262695, 26.21069336, 26.59350586, 27.02416992, 27.0480957 , 34.05834961, 54.06030273, 60.87915039, 67.02807617, 67.91333008, 72.96166992, 73.6315918 , 74.06225586, 75.09106445, 75.5456543 , 75.85668945, 77.00512695, 77.81860352, 80.6418457 , 81.09643555, 81.4074707 , 81.90991211, 88.10668945, 88.51342773]

gregghallinan, 2023-02-09:
Hi Casey,

Is this day time? The figure does not look like RFI to me, but rather ADC saturation.

The frequencies below look reasonable,  so my comment is with respect to the figure only.

Best,
Gregg

[Image]<https://user-images.githubusercontent.com/1878317/217683384-78b94eef-76ac-4e5e-8e32-11636011ab80.png>

caseyjlaw, 2023-02-13:
Good point. This was a day time observation. We'll run this again with the new attenuation settings.

caseyjlaw, 2023-02-17:
Test vector work looks good.
https://github.com/ovro-lwa/mnc_python/pull/10#issuecomment-1434377728

caseyjlaw, 2023-02-22:
LA DWP beacon at 48.36 MHz could also be used.

closed 2023-02-22 by caseyjlaw