ovro-lwa/lwa-issues #177: Test bifrost beamformer with random data/weights

caseyjlaw, 2022-10-25:
Verifying output matches a simple python matrix multiply operation.
This test uses the control library to set weights.

iveydavis, 2022-11-02:
Manipulating the calibration tables and confirming the output

jack-h, 2022-11-02:
If you want all inputs to the beamformer to be constant and known, you could use the F-engine "ramp" test mode (https://raw.githubusercontent.com/realtimeradio/caltech-lwa/master/docs/build/latex/LWA352SNAP2F-Engine.pdf)

jack-h, 2022-11-02:
`write_freq_ramp()`
`tvg_enable()`

[image]

iveydavis, 2022-11-08:
Starting the x-engine pipelines won't overwrite the use of f-engine test vectors will it? And Is there a reason to use the test vector block instead of injecting noise or using the dummy source functionality of bifrost?

jack-h, 2022-11-09:
Correct, starting the X-engine pipelines won't affect what the F-engines
are doing.
The main benefit of using the test vectors on the F-engines is that it's
easy, and you know exactly what is going into the B/X pipelines. Generating
data in bifrost would also work, but tests less of the pipeline. Testing
with noise would be good too, it's just a little tricky to know what 4-bit
data ends up going to the X/B engines after FFT / requantization processing

caseyjlaw, 2023-02-15:
Ran the beamformer_sum_test.py and beamformer_test.py on gpu09 and am consistently getting "Match OK". Also suitable for this issue.

closed 2023-02-15 by caseyjlaw