ovro-lwa/lwa-issues #178: Test x-eng bifrost beamforming blocks

caseyjlaw, 2022-10-25:
Include the `Beamform` and `BeamformSumBeams` blocks in the test pipeline, and compare to a Python implementation.

iveydavis, 2022-11-02:
there are scripts related to this on gpu0X under /proj/lwa-shell/caltech-bifrost-dsp/pipeline/verification/tests/beamformer

caseyjlaw, 2022-11-30:
Need bifrost on cal-im to do this. `datarecorder` environment should have it.

iveydavis, 2023-02-01:
I tried running  beamformer_sum_test.py in the directory /home/pipeline/proj/lwa-shell/caltech-bifrost-dsp/pipeline/verification/tests/beamformer that looks like it should be addressing this question. I was on the pipeline user on calim00 and in the datarecorder cenvironment. This is the output:

```
datarecorder) [pipeline@lwacalim00 beamformer]$ python beamformer_sum_test.py
Generating test data
Writing test data to file
Writing testfile: /tmp/temp_beam_testfile.dat
Closing file
ring_impl.cpp:105 Condition failed: space==BF_SPACE_SYSTEM
ring_impl.cpp:105 error 10: BF_STATUS_INVALID_ARGUMENT
ring.cpp:41 error 10: BF_STATUS_INVALID_ARGUMENT
Traceback (most recent call last):
  File "beamformer_sum_test.py", line 190, in <module>
    main()
  File "beamformer_sum_test.py", line 138, in main
    input_ring = Ring(name="input", space="cuda_host")
  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/ring.py", line 61, in __init__
    self, _bf.bfRingCreate, _bf.bfRingDestroy, name, space)
  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/libbifrost.py", line 50, in __init__
    _check(constructor(ctypes.byref(self.obj), *args))
  File "/home/pipeline/.conda/envs/datarecorder/lib/python3.6/site-packages/bifrost-0.9.0-py3.6.egg/bifrost/libbifrost.py", line 111, in _check
    raise RuntimeError(status_str)
RuntimeError: b'BF_STATUS_INVALID_ARGUMENT'```

I don't understand what role/ functionality the Ring class has well enough to understand how to fix this.

jaycedowell, 2023-02-01:
This looks like a Bifrost build problem where it was built without CUDA support.  A memory space of `cuda_host` was requested by the script but only `system` is supported.

caseyjlaw, 2023-02-01:
Do you think that using `system` memory is sufficient for this test or do we need to recompile?

jaycedowell, 2023-02-01:
You'll probably need to recompile since I think the beamforming blocks run on the GPU.

yupinghuang, 2023-02-02:
errrr....should these tests run on the GPU node where the beamformer runs?

iveydavis, 2023-02-08:
Ran the beamformer_sum_test.py and beamformer_test.py on gpu09 and am consistently getting "Match OK".   I'm going to take that to mean that things are still working okay and I'll close out this issue and I believe issue 177 as well