ovro-lwa/lwa-issues #183: Beam pointing check

caseyjlaw, 2022-10-25:
Point several beams next to each other along the track of a bright source close to transit to observe the source pass through the beams.

nmahesh1412, 2022-11-08:
An initial effort towards this was done with the following 3 beams in the sky:
1.) con.start_xengine_bf(1, targetname='cyg a', track=False, calibrate=True)

As the source drifts out of beam01, there was another beam set on the source again as:
2.) con.start_xengine_bf(2, targetname='cyg a', track=False, calibrate=True)

And for reference, we had a third beam as an off-beam 
3.) con.start_xengine_bf(3, targetname='zenith', track=False, calibrate=True)

nmahesh1412, 2022-11-08:
Some plots:

[image]

nmahesh1412, 2023-02-08:
tests done on Nov 8th:
On Cas A:

 

Test 1

Started beamformer 1 with calibration, no tracking

Started beamformer 2 with calibration (5 minutes later), no tracking

Started dr1, dr2

/data01/beam01/059892_022412001025902cc6d

/data01/beam02/059892_022412009342902f6fa

Hypothesis: This should be an off-beam/on-beam test. The first one was a few minutes earlier than the second, which started on target. Second should start higher than the first at its start, since it was pointed on target when recording started.

 

Test 2

Started beamformer 1 and 2 with calibration, no tracking

Started dr1, dr2

/data01/beam01/059892_02310499641786581bc

/data01/beam02/059892_0231050120588659583

(lots of missing packets)

Hypothesis: Both start on target and let source drift out, so should be similar. Starts high, goes low.

 

Test 3

Start beamformer 1 tracking (leave beamformer 2 off target)

/data01/beam01/059892_023326345737da982aa

/data01/beam02/059892_023326354834da99b12

Hypothesis: This should be an off-/on-beam test. Beam 2 is off somewhere from prior test. Beam 1 is on and tracking. Should be high and stay high.

nmahesh1412, 2023-02-08:
import numpy as np
import matplotlib.pyplot as plt
from astropy.time import Time
from astropy import units as u
from astropy.coordinates import Angle
from astropy import coordinates as coord
from astropy.coordinates import EarthLocation
from astropy.coordinates import SkyCoord

def altaz_sources(lat,lon,coos,times):
    """
    Transform the source coordinates from the absolute sky frame to the antenna frame.
    """
      
    antenna_loc = EarthLocation.from_geodetic(lat =lat, lon = lon)
    aa_frame = coord.AltAz(obstime=times, location=antenna_loc)
    
    aa_coos = coos.transform_to(aa_frame)  
    
    return aa_coos.alt.deg, aa_coos.az.deg

cygA_ra = Angle('19 59 28.35645829',unit=u.hourangle)
cygA_dec = Angle('+40 44 02.0966496', unit=u.degree)

casA_ra = Angle('23 23 24.000',unit=u.hourangle)
casA_dec = Angle('+58 48 54.00', unit=u.degree)

tauA_ra = Angle('05 34 31.94',unit=u.hourangle)
tauA_dec = Angle('+22 00 52.2', unit=u.degree)

sgrA_ra = Angle('17 45 40.03599',unit=u.hourangle)
sgrA_dec = Angle('-29 00 28.1699', unit=u.degree)

ovro_lat = 37.37
ovro_lon = -118.37

objects = [cyg_A,cas_A,tau_A,sgr_A, c295, c286]

off = np.linspace(-12,12,48)*u.hour
alt = np.zeros((len(objects),len(off)))
az = np.zeros_like(alt)
for j in range(len(objects)):
    for i in range(len(off)):
        alt[j,i],az[j,i] = altaz_sources(ovro_lat,ovro_lon,objects[j],Time.now()+off[i])
    

new = Time.now()+off
label = ['cyg_A','cas_A','tau_A','sgr_A','3c295','3c286']
plt.figure(figsize=(12,5))
plt.subplot(2,1,1)
plt.title('Source positions relative to OVRO (+37.27, -118.37)')
plt.subplots_adjust(hspace=0.5)
for i in range(len(objects)):
    plt.plot(new.value, alt[i],label=label[i])
plt.grid()
plt.ylabel('elevation\n(deg)')
plt.legend()
plt.subplot(2,1,2)
for i in range(len(objects)):
    plt.plot(new.value, az[i])
plt.grid()
plt.ylabel('azimuth\n(deg)')
plt.xlabel('Date-time (UTC)')

caseyjlaw, 2023-03-15:
24 hour run sees the sun, but not definitive (also in off beam).
Pulsar test not detecting pulsar, so more work needed.

caseyjlaw, 2023-03-29:
Pulsar detected!