ovro-lwa/lwa-issues #188: Test t-eng spectral properties

caseyjlaw, 2022-10-25:
Confirm that spectrum order is correct at each stage through to final recorded data. 
Spur checks -- Confirm that no unexpected spikes occur in the passbands.

jaycedowell, 2022-10-27:
#116 works towards this some since it focuses on extracting the WWV second ticks from the 10 MHz signal.  Those data are from the second day of the recent moon bounce experiment (MJD 59872).

jaycedowell, 2022-10-27:
There was also at least one successful pulsar observation with the T-engine.  We might want to repeat that during a better interference environment.  A fringe test with LWA1/LWA-SV would also be nice.

caseyjlaw, 2023-05-03:
Marin also saw 1-channel offset between power/voltage beams.

jaycedowell, 2023-05-03:
https://github.com/lwa-project/ovro_data_recorder/issues/30

jaycedowell, 2023-12-07:
Gain settings are working.

jaycedowell, 2023-12-18:
Everything seems ok modulo https://github.com/ovro-lwa/lwa-issues/issues/460.

closed 2023-12-18 by jaycedowell