ovro-lwa/lwa-issues #189: Reproduce power beam from voltage beam

caseyjlaw, 2022-10-25:
Record voltage beam and verify that power beam can be generated from time beam data product.

jaycedowell, 2022-10-27:
I'm looking at this using data from the third day of the moon bounce experiment (MJD 59873) where I have simultaneous power and voltage beam data.  For the power beam data I focus on the ~10 +/- 5 MHz region where the voltage beam was tuned to and look at the first 200 s of data.

Processing the voltage beam data into spectra with a similar temporal resolution (~0.8 ms vs ~1 ms) things look ok:
[image]
and there is qualitatively good correspondence between the two.

If I integrate over frequencies and plot power as a function of time I see:
[image]
There is a time offset here between the voltage beam (blue) and the power beam (orange) of about 160 ms.  If I adjust the power beam by this amount you get the green line and that agrees with the blue one.

To figure out which time is correct I looked at the 10 MHz WWV signal as in #116:
[image]
In this plot "Full Res. Voltage" refers to a processed voltage beam data product with ~2 kHz spectral resolution at the same ~0.8 ms time resolution.  The peak in the "full res." data is close to the expected time of the second tick (the vertical dotted line) and there is a corresponding peak, albeit at lower SNR, for the blue line.  The second tick also shows up in the power beam (orange) at the same significance as in the blue but about 160 ms later.  At this point I'm not really sure where this 160 ms lag in the power beam comes from.

jaycedowell, 2022-10-28:
I looked at the data from MJD 59872 and I seem something similar in the integrate power as a function of time:
[image]
The offset in this case is about ~-400 ms.  400 ms is probably a larger error than you can create in the X-engine pipelines so I think this is something with the power beam recorders.

jaycedowell, 2022-10-28:
Here's a view of the data from MJD 59872 where they have been converted to pseudo-Stokes parameters.
[image]

jaycedowell, 2022-12-22:
I've looked at this again over the last couple of days and I still haven't been able to isolate where things are going wrong.  On the data recorder side I have looked at:
 * time inaccuracy with `LWATime` and the conversion of timetags, 
 * problems keeping track of time within the data recorder, and
 * conversion problems translating a `LWATime` to the tuple stored in the HDF5 file.

One thing I did notice in this last round of testing is that the offset seems to change with restarts of the x-engine pipelines.  It also _may_ drift so that the offset increases with increasing time since the last pipeline restart.  I'm not 100% sure on that last point.

So the clues are:
 1) Restarting the power beam stream changes the offset.
 2) Whatever is causing it needs to not impact the timing of the voltage beam.
 3) The offset _may_ increase the longer the power beam stream is running.

Taking those together it could still be either the x-engines or the data recorder.  (2) really limits where in the x-engine the problem could be, though.  It would be limited to the power beam accumulator and packetizers.  It could also be hiding in some part of the data recorder that I haven't considered.

jaycedowell, 2023-03-29:
@GregBTaylor and I were talking about this.  It seems like what is needed is a series of test to better determine where this comes from.  Something like:
 1. Run the voltage beam and 2/3/4 power beams all pointing at the same (fixed?) target to see if (a) there is an offset between the power beams, (b) if there is an offset between each power beam and the voltage beam, and (c) if (a) and (b) are consistent with each other.
 2. Repeat the test in (1) with multiple recordings _without_ restarting either the data recorders or the x-engines.
 3. Repeat (1) and possibly (2) with restarting _only_ the data recorders between each set of tests.
 4. Repeat (1) and possibly (2) with restarting _only_ the x-engines between each set of tests.

caseyjlaw, 2023-03-29:
Sounds like a fairly simple test. For bonus points, we should set this up with a SDF. I'll try to learn how to do that today from Ivey.

caseyjlaw, 2023-05-16:
For book keeping purposes, I'll list the other time anomaly issues:
#283 
#225 
#116

jaycedowell, 2023-12-01:
It's been a while so I've taken another look at this.  From my tests today for a zenith pointing and comparing against the voltage beam the offsets are:
 * Beam 1: 360 ms
 * Beam 3: -400 ms
 * Beam 4: -400 ms

Interesting that beams 3 and 4 have the same offset.  After restarting dr-beam-4 I see:
 * Beam 1: 360 ms
 * Beam 3: -400 ms
 * Beam 4: 0 ms

That causes a change in beam 4.  I did not expect that.  After resetting the destination address/port for beam 3:
 * Beam 1: 360 ms
 * Beam 3: -400 ms
 * Beam 4: 0 ms

Same as before.  Restarting dr-beam-1 now gives:
 * Beam 1: -490 ms
 * Beam 3: -400 ms
 * Beam 4: 0 ms

So it seems to follow the data recorder but I can't think of what the mechanism would be.  For good measure, another restart of dr-beam-1 gives:
 * Beam 1: -400 ms
 * Beam 3: -400 ms
 * Beam 4: 0 ms

jaycedowell, 2023-12-04:
https://github.com/lwa-project/ovro_data_recorder/commit/982333f868b1e94c92272042585d21ef90d84172 helps and the offset is now ~1 ms, slightly longer than the native power beam temporal resolution.

jaycedowell, 2023-12-04:
This ~1 ms error needs a bifrost patch (https://github.com/realtimeradio/bifrost/pull/3).  This gets a little tricky because I don't know where bifrost on calim comes from.  Is it still from my `/home/pipeline/bifrost` directory?  That's at least what I used for testing this patch.

caseyjlaw, 2023-12-05:
At least the gpu servers get it from the bifrost submodule in the [x-engine code](https://github.com/realtimeradio/caltech-bifrost-dsp/blob/main/deploy). The bifrost module is separately available in `lwa-shell`, which is how Rick does most deployment. @rh-codebase do you know how bifrost is installed on cal-im?

jaycedowell, 2023-12-07:
The version of bifrost in `lwa-shell` wouldn't be the right one since it points to the original ledatelescope version.  What's in the bifrost sub-module from caltech-bifrost-dsp is closer but it needs a little tweaking for calim (no xgpu or CUDA support; that unmerged patch I mentioned earlier).

caseyjlaw, 2023-12-08:
Would you suggest we change the remote for the bifrost in `lwa-shell` and then you can do a PR to add a deployment script?

jaycedowell, 2023-12-08:
I'm not sure.  Are any of the other packages using bifrost or it is only ovro_data_recorders and caltech-bifrost-dsp?  If it's only those two it might make sense drop bifrost from `lwa-shell` and add a submodule for bifrost into ovro_data_recorders.

caseyjlaw, 2023-12-08:
This would only be for ovro_data_recorders, so a submodule would make sense. You could fix it to a particular commit/version too.

jaycedowell, 2023-12-14:
`main` currently as a bifrost submodule in it and a new `deploy` script that will install this version of bifrost.  It seems to work on the T-engine but I haven't tried it on calim.  One thing to note about it is that it currently doesn't try to build CUDA support on calim.  Maybe you want to change that based on @nkosogor's beamformer work?

caseyjlaw, 2023-12-14:
We removed the `--disable-cuda` and got `deploy` to work with CUDA support. Could you make that change to the script and we can restart the recorders to close this out?

jaycedowell, 2023-12-14:
It's in - https://github.com/lwa-project/ovro_data_recorder/commit/5e9dd0532d7cbf4c37a880c19b93003e7b0bfce4.

When did you want to do the restart?

caseyjlaw, 2023-12-14:
I think now (late afternoon) is a good time. Are you free now?

jaycedowell, 2023-12-14:
Sure.  I've started `deploy` on calim01.

jaycedowell, 2023-12-15:
Everything seems ok.  For the record the bifrost submodule currently points to https://github.com/realtimeradio/bifrost/commit/b739e3d2558cf1783952230c24ff796b7e93438e which is on the https://github.com/realtimeradio/bifrost/tree/lwa352-myrepos branch.

closed 2024-01-17 by caseyjlaw