ovro-lwa/lwa-issues #190: Verify voltage buffer recording

caseyjlaw, 2022-10-25:
Basic signal path verification -- Confirm signals are transmitted through system, and that all subsystems are properly connected. 
Record data and verify structure on disk.

jack-h, 2022-11-01:
So, triggered recording currently seems completely non-functional 🤦 
I'm trying to figure out why this is, but my current theory is that it is related to the ZFS filesystem on the NVMe drives. The system was only previously tested with XFS.

jack-h, 2022-11-01:
See issue #200

jack-h, 2023-01-25:
- Recording buffer file functionality now tested on gpu-02.
- Working on scripting a test to verifying file contents
- Other GPU nodes (all but 2) need re-provisioning to roll-out filesystem changes across all systems

caseyjlaw, 2023-04-12:
Nikita and I have successfully run code to trigger voltage recording on gpu02:/data1. We are ready to test triggering across all gpu nodes/disks.
Can this reprovisioning work proceed, if we agree to suspend observing for a day?

jack-h, 2023-04-13:
Fab! I'm 1000% in favour of firing up the reprovisioning of GPU1-8 (I guess skipping 2 since it's done already), as soon as @rh-codebase is able to do it.

caseyjlaw, 2023-08-30:
This is working in principle and issues will be tracked elsewhere.

closed 2023-08-30 by caseyjlaw