ovro-lwa/lwa-issues #199: SNAP behavior when optical FSP module unplugged and replugged.

rh-codebase, 2022-10-28:
Mark worked on SNAP3 which included unplugging and plugging the optical fiber FSP on the board. Afterwards, no spectra was seen from the SNAP. The board still reported as being programmed. Procedure used to get the spectra back was to follow the reprogramming section of the System Use Recipe:

>>> from lwa_f import snap2_feng_etcd_client 
>>> ec = snap2_feng_etcd_client.Snap2FengineEtcdControl() 

First check whether board was programmed:
>>> ec.send_command(0, 'fpga', 'is_programmed', n_response_expected=11) 
{10: True, 9: True, 6: True, 5: True, 8: True, 7: True, 3: True, 1: True, 4: True, 2: True, 11: True}
>>> FPGFILE = '/home/ubuntu/proj/lwa-shell/caltech-lwa/snap2_f_200msps_64i_4096c/outputs/snap2_f_200msps_64i_4096c_2022-04-14_0450.fpg'

Start reprogramming steps:

 ec.send_command(0, 'feng', 'program', kwargs={'fpgfile':FPGFILE},  timeout=60*7, n_response_expected=11) 
{8: None, 3: None, 1: None, 4: None, 2: None, 7: None, 5: None, 6: None, 9: None, 10: None, 11: None}
>>> ec.send_command(0, 'feng', 'initialize', kwargs={'read_only':False},  timeout=300, n_response_expected=11) 
{10: None, 2: None, 4: None, 11: None, 8: None, 3: None, 5: None, 7: None, 6: None, 9: None, 1: None}
>>> CONFIGFILE = '/home/ubuntu/proj/lwa-shell/caltech-lwa/control_sw/config/lwa_corr_config.yaml'

>>> for i in range(1,12):
...   print(i)
...   ec.send_command(i, 'feng', 'cold_start_from_config', kwargs={'config_file':CONFIGFILE, 'program':False, 'initialize':True}, timeout=20) 
... 
1
2
Command status from FID 2 returned: 'error'
{}
3
4
Command status from FID 4 returned: 'error'
{}
5
Command status from FID 5 returned: 'error'
{}
6
7
8
9
10
{}
11
Command status from FID 11 returned: 'error'
{}
>>> ec.send_command(0, 'fpga', 'is_programmed', n_response_expected=11) 
{5: True, 9: True, 10: True, 3: True, 11: True, 6: True, 7: True, 4: True, 8: True, 2: True, 1: True}

Are the errors expected? What steps are needed when they occur?

Despite the above errors, spectra plots were obtained by running:

 python /data1/rfisurvey/scripts/genpdf_etcd_tmp.py

jack-h, 2022-10-28:
> Assigned #199 <https://github.com/ovro-lwa/lwa-issues/issues/199> to
> @jack-h <https://github.com/jack-h>.

This is expected, and probably healed with
`CosmicFengine.eth.reset()` followed by `CosmicFengine.eth.enable_tx()`

jack-h, 2022-10-28:
The errors aren't expected, but to figure out what they are referring to would require checking the logs. In any case, they're "just" initialization issues, and have nothing to do with the SFP

rh-codebase, 2022-10-28:
@Princemongo :  Commands to run after fiddling with the SFP modules on the SNAPS:

`CosmicFengine.eth.reset()` followed by `CosmicFengine.eth.enable_tx()`

I'll check logs regarding the Errors and file separate issue.

closed 2022-10-28 by rh-codebase