ovro-lwa/lwa-issues #200: GPU nodes -- incorrect NVMe FS layout

jack-h, 2022-11-01:
OS should be installed on pcie d9 (= `/dev/nvme0` on `gpu01`, but I don't know if this is consistent over nodes) so that the 2 sets of 4 Samsung NVMes (pcie 5e-61, and 86-89) can serve as the two triggered buffer data stores.

Separately, I'm having issues writing data with the `DIRECT_IO` flag to the /data0 or /data1 partitions. This works on the test system with `xfs` filesystems. I'm wondering if the issue here might be `zfs` related?

jack-h, 2022-11-01:
I can confirm that trashing the ZFS and making an MDADM raid 0 of NVMes 86-89 allows recording

jack-h, 2022-11-01:
@rh-codebase Any chance you can reconfigure / reimage?

rh-codebase, 2022-11-03:
The preferred solution which has worked in the past:

To find zfs pools:
: sudo zpool import
Then import them using their name. i.e.
: sudo zpool import data0
: sudo zpool import data1
: sudo zpool status

It looks like 'trashing the ZFS' has completely removed the pool info from the drives so the above cmds don't work.

jack-h, 2022-11-03:
Maybe I'm misunderstanding, but I'm not sure what the above is a solution to. Until I removed ZFS and replaced with XFS the DIRECT_IO code wouldn't run.

rh-codebase, 2022-11-03:
my misunderstanding. I thought the zfs mounts didn't exist. I also wasn't aware that DIRECTIO was a requirement. This option is  definitely filesystem dependent. Here's a summary from:  [ryao](https://github.com/ryao) commented [on Jul 23, 2015](https://github.com/openzfs/zfs/issues/224#issuecomment-124153060)

As such. it looks like XFS will be a requirement for now.  @jack-h send me an e-mail on any needed options when making the XFS filesystem on the backed mdadm raid0.

rh-codebase, 2022-11-03:
@jack-h Disregard last request. history on gpu01 gave me the needed info.

rh-codebase, 2023-06-09:
should all be fixed with new OS install on Intel disks and mdamd setup on /data0 /data1 with XFS filesystems.

closed 2023-06-09 by rh-codebase