ovro-lwa/lwa-issues #201: lwautils api using numbering 0-15 while rfi spectra are 1-16

VinAstro, 2022-11-01:
Made several mistakes now while trying to debug the system by using the 1-16 numbering system when using Rick's lwautils api. Can we make this a standard where its either 0-15 or 1-16? I just wasted an hour figuring out what I was doing wrong because of this.

caseyjlaw, 2022-11-01:
That sounds frustrating! This could be a good topic for the monitoring sprint.
Just to be clear, is this a difference between the ARX numbering and the antenna numbering?

VinAstro, 2022-11-01:
The numbering system for the ARX channels when using Rick's API when changing the ARX settings is from 0-15. But physically and on the system configuration spreadsheet its 1-16. So if I want to turn channel 5(as per the system config) off, I need to call number 4 on the API. This is the confusion I was getting myself into.

rh-codebase, 2023-06-09:
the python arx library's API is meant to be used by applications not interactively by humans. Translating natural 0-index based numbering of the programming language to human likable numbers is the job of an application layer. One such application will be the WebUI but the ARX feature is still WIP.

closed 2023-06-09 by rh-codebase