ovro-lwa/lwa-issues #202: data recorder thread dying

yupinghuang, 2022-11-15:
```
In [10]: con.status_dr(recorders=['drvs'])
Out[10]: 
['recording',
 'WARNING: drvs not fully operational: 16 sub-bands error: drvs7601=Only 9 of 10 threads active - MainThread,etcd3_watch_7f396f908860,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7602=Only 9 of 10 threads active - MainThread,etcd3_watch_7eff90be2898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7701=Only 9 of 10 threads active - MainThread,etcd3_watch_7fd5e340a860,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7702=Only 9 of 10 threads active - MainThread,etcd3_watch_7f9376efb898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7801=Only 9 of 10 threads active - MainThread,etcd3_watch_7f2d95bfb8d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7802=Only 9 of 10 threads active - MainThread,etcd3_watch_7f2fe7cf9898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7901=Only 9 of 10 threads active - MainThread,etcd3_watch_7fed7b1368d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs7902=Only 9 of 10 threads active - MainThread,etcd3_watch_7fd0eaf94828,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8001=Only 9 of 10 threads active - MainThread,etcd3_watch_7f2783b11860,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8002=Only 9 of 10 threads active - MainThread,etcd3_watch_7f155b6d7898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8101=Only 9 of 10 threads active - MainThread,etcd3_watch_7f28aa3ba8d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8102=Only 9 of 10 threads active - MainThread,etcd3_watch_7ff3b01a7898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8201=Only 9 of 10 threads active - MainThread,etcd3_watch_7f5f8db158d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8202=Only 9 of 10 threads active - MainThread,etcd3_watch_7f234b477898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8301=Only 9 of 10 threads active - MainThread,etcd3_watch_7f4d9c38e8d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8302=Only 9 of 10 threads active - MainThread,etcd3_watch_7f5b0551a898,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger']
```

restarting the dr services seem to fix things

jaycedowell, 2022-11-28:
I've seen this before where the `WriterOp` thread dies but I'm not sure what causes it.  Did you find the data recorders in this state?

yupinghuang, 2022-11-28:
I did. It hasn't happened since.

jaycedowell, 2022-11-28:
Looking through the log from the 14th it looks like this error condition first appeared after a stop command was sent to the recorders.  Stop is processed by `WriterOp` so that might have triggered this.

caseyjlaw, 2022-12-19:
I'm seeing this state again this morning.

Looking at the log /data01/log/dr-vslow-1.lwacalim01.log, I see:
```
2022-12-19 01:50:42 [DEBUG   ] === Status Report ===
2022-12-19 01:50:42 [DEBUG   ]  summary: normal
2022-12-19 01:50:42 [DEBUG   ]  info: System operating normally
2022-12-19 01:50:42 [DEBUG   ]  queue size: 0
2022-12-19 01:50:42 [DEBUG   ]  active operation: False
2022-12-19 01:50:42 [DEBUG   ] ===   ===
2022-12-19 01:50:52 [DEBUG   ] === Performance Report ===
2022-12-19 01:50:52 [DEBUG   ]  max acquire/process/reserve times: 10.007/10.007/0.000
2022-12-19 01:50:52 [DEBUG   ]  receive data rate: 19490054.697 B/s
2022-12-19 01:50:52 [DEBUG   ]  missing data fraction: 0.000%
2022-12-19 01:50:52 [DEBUG   ]  pipeline lag: 31.212495
2022-12-19 01:50:52 [DEBUG   ]  load average: 16.16, 15.38, 16.39
2022-12-19 01:50:52 [DEBUG   ] ===   ===
2022-12-19 01:51:03 [DEBUG   ] === Performance Report ===
2022-12-19 01:51:03 [DEBUG   ]  max acquire/process/reserve times: 0.001/14.359/0.000
2022-12-19 01:51:03 [DEBUG   ]  receive data rate: 37949105.967 B/s
2022-12-19 01:51:03 [DEBUG   ]  missing data fraction: 0.000%
2022-12-19 01:51:03 [DEBUG   ]  pipeline lag: 31.212495
2022-12-19 01:51:03 [DEBUG   ]  load average: 16.16, 15.42, 16.39
2022-12-19 01:51:03 [DEBUG   ] ===   ===
2022-12-19 01:51:03 [DEBUG   ] === Status Report ===
2022-12-19 01:51:03 [DEBUG   ]  summary: error
2022-12-19 01:51:03 [DEBUG   ]  info: Max. processing time at >125% of gulp (14.359 s = 143.1% )
2022-12-19 01:51:03 [DEBUG   ]  queue size: 0
2022-12-19 01:51:03 [DEBUG   ]  active operation: False
2022-12-19 01:51:03 [DEBUG   ] ===   ===
2022-12-19 01:51:13 [DEBUG   ] === Performance Report ===
2022-12-19 01:51:13 [DEBUG   ]  max acquire/process/reserve times: 0.001/11.262/0.000
2022-12-19 01:51:13 [DEBUG   ]  receive data rate: 75682117.886 B/s
2022-12-19 01:51:13 [DEBUG   ]  missing data fraction: 0.000%
2022-12-19 01:51:13 [DEBUG   ]  pipeline lag: 31.212495
2022-12-19 01:51:13 [DEBUG   ]  load average: 16.13, 15.44, 16.38
2022-12-19 01:51:13 [DEBUG   ] ===   ===
2022-12-19 01:51:23 [DEBUG   ] === Performance Report ===
2022-12-19 01:51:23 [DEBUG   ]  max acquire/process/reserve times: 0.001/11.263/0.000
2022-12-19 01:51:23 [DEBUG   ]  receive data rate: 0.000 B/s
2022-12-19 01:51:23 [DEBUG   ]  missing data fraction: 0.000%
2022-12-19 01:51:23 [DEBUG   ]  pipeline lag: 31.212495
2022-12-19 01:51:23 [DEBUG   ]  load average: 14.87, 15.19, 16.29
2022-12-19 01:51:23 [DEBUG   ] ===   ===

```

jaycedowell, 2022-12-19:
This looks like an excerpt from a different error condition but, yes, about 7:30 on the 19th there was this condition again.  Is there anything syslog from the data recorder around that time?

caseyjlaw, 2022-12-19:
Not clear to me. I can't access older logs for some reason. I'll try to resolve that in another issue.

jaycedowell, 2022-12-19:
Great, thanks.

jaycedowell, 2022-12-22:
I pushed a couple of changes onto calim last night that should address this.

caseyjlaw, 2023-04-11:
I saw this failure mode again today, but with a different message:
`Found 1 missing threads - WriterOp, No recording space (100.0% used)`

This happened after I restarted all data recorders due to an odd state that I did not understand. Could be prior and after restart, that there was a disk space issue (or issue with measuring it).
Do either of you know how the recorders measure disk space?

jaycedowell, 2023-04-11:
Having no space sounds like a good reason for the `WriterOp` to die.  What was the "odd state"?

The disk space is found by looking at all objects in the directory being monitored and calling `stat` on them to get a size.  If the object is a directory it recurs down inside it to get a total size that includes its contents.

caseyjlaw, 2023-04-11:
Actually, yes, you are right. There is just no disk space on that server.
Looks like the slow visibilities were being recorded all day. Others servers have a bit free, but not the one with the error.

jaycedowell, 2023-04-11:
I would have thought that the quota management would have done something to prevent that from happening.  However, that only tracks what happens within the directory it monitors.  You could get in a situation where other users of a disk could force an out of space condition before the quota management would think there was a problem based on its limited scope.

yupinghuang, 2023-04-11:
Jayce is right. on calim04 the data recorder borked at a 6.4 TB disk usage, when the quota is set to 8TB

Between storing the 24 hour run and the solar probe run still on the rest of the nodes this is a genuine running out of disk space situation. I'm gonna shrink the data recorder quota to 2TB (from 8 TB).

caseyjlaw, 2023-06-06:
I saw this error again (for recorder in lwacalim03):
`drvs7801=Found 1 missing threads - WriterOp`. A restart resolved it.

jaycedowell, 2023-06-06:
I looked through dr-vslow-5.lwacalim03.syslog to see what the underlying problem for this latest one was.  It was a `RuntimeError: File '/data03/slow/20230606_123243_45MHz.ms' already exists` thrown when trying to create a new measurement set.  I'm not sure how the data recorder can get into a situation where it wants to write the same timestamp twice.

caseyjlaw, 2023-06-29:
I'm seeing a similar error from another part of the data recorder:
`Found 1 missing threads - StorageLogger`.

caseyjlaw, 2023-06-29:
Hmm, I'm also seeing the original error for this issue: `Found 1 missing threads - WriterOp`.
I had just restarted all recorders to clear the StorageLogger issue.

caseyjlaw, 2023-10-05:
Another instance for power beam 2:
```
2023-10-04 22:57:15 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 22:57:33 [INFO    ] Ended operation - <HDF5Writer filename='/data01/beam02/060221_2227300000002ae5729',
    start_time='2023-10-04 22:27:30', stop_time='2023-10-04 22:57:30',
    reduction=<ReductionOperation time_avg=64, chan_avg=1,
    pols='XX,YY,CR,CI'>>
2023-10-04 22:57:35 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 22:57:44 [INFO    ] HDF5Record - Scheduled recording for 2023-10-04 22:57:59.999000 to 2023-10-04 23:27:59.999000 to /data01/beam02/060221
_2257599990006ddfab9
2023-10-04 22:57:56 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 22:58:01 [INFO    ] Started operation - <HDF5Writer filename='/data01/beam02/060221_2257599990006ddfab9',
    start_time='2023-10-04 22:57:59.999000', stop_time='2023-10-04
    23:27:59.999000', reduction=<ReductionOperation time_avg=64,
    chan_avg=1, pols='XX,YY,CR,CI'>>
2023-10-04 22:58:17 [ERROR   ] Status report: No recording space (99.3% used), Max. processing time at >125% of gulp (4.247 s = 413.4% )
2023-10-04 22:58:37 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 22:58:59 [ERROR   ] Status report: No recording space (99.3% used), Max. processing time at >125% of gulp (4.451 s = 433.3% )
2023-10-04 22:59:22 [ERROR   ] Status report: No recording space (99.3% used), Packet loss during receive >10% (14.4% missing)
2023-10-04 22:59:44 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 23:00:08 [ERROR   ] Status report: No recording space (99.3% used)
...
and a bit later
...
2023-10-04 23:23:58 [ERROR   ] Status report: No recording space (99.3% used), Max. processing time at >125% of gulp (1.466 s = 142.7% )
2023-10-04 23:24:17 [ERROR   ] Status report: No recording space (99.3% used), Max. processing time at >125% of gulp (1.383 s = 134.6% )
2023-10-04 23:24:40 [ERROR   ] Status report: No recording space (99.3% used)
2023-10-04 23:25:34 [ERROR   ] Status report: Found 1 missing threads - WriterOp, No recording space (99.3% used)
2023-10-04 23:25:54 [ERROR   ] Status report: Found 1 missing threads - WriterOp, No recording space (99.3% used), Packet loss during receive >10% (13.3% missing)
2023-10-04 23:26:14 [ERROR   ] Status report: Found 1 missing threads - WriterOp, No recording space (99.3% used)
```

It remained in that state until the next day when I restarted.

jaycedowell, 2023-10-05:
Those all look like good reasons for the WriterOp to die.

jaycedowell, 2023-11-14:
This should be addressed with https://github.com/lwa-project/ovro_data_recorder/commit/36d898980f9eb0f7cfe7e10b77f4115174265099 and https://github.com/lwa-project/ovro_data_recorder/commit/35e34d57daec61863011edc2c051cf0ca3b1c075.  If WriterOp runs into a problem it will be logged but it should to continue to iterate through incoming data.

closed 2023-11-14 by jaycedowell