ovro-lwa/lwa-issues #203: con.stop_dr() leaves data recorder in wonky state?

yupinghuang, 2022-11-15:
More to come tomorrow morning but here's what happened. on calim00
```
In [45]: con.stop_dr(recorders=['drvs'])
recording on drvs stopped
```
about five minutes later
```
In [61]: con.status_dr(recorders=['drvs'])
Out[61]: 
['2 recording; 14 idle',
 'WARNING: drvs not fully operational: 2 sub-bands have not updated in 120 s: drvs8101,drvs8102']
```

yupinghuang, 2022-11-15:
About 9 hours later
```
In [3]: con.status_dr(recorders=['drvs'])
Out[3]: 
['1 recording; 15 idle',
 'WARNING: drvs not fully operational: 1 sub-bands have not updated in 120 s: drvs8102']
```

caseyjlaw, 2022-11-22:
Maybe related: I tried to start slow visibility data recording and simple (mnc-python) logging/status showed it was working. However, no data was written.
Restarting data recorder services, then restarting observing steps fixed this.

yupinghuang, 2022-11-23:
I just realized that calim07 used to record 70 & 75 MHz but now it is recording 80 & 85 MHz, possibly because I started the controller the following way when I tried to skip gpu02 since it is being reprovisioned.
```
In [12]: con = control.Controller("/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml",
    ...:                          xhosts=["lxdlwagpu01", "lxdlwagpu03", "lxdlwagpu04", "lxdlwagpu05", "lxdlwagpu06", "lxdlwagpu07", "lxdlwagpu08"])
```

yupinghuang, 2022-11-23:
Maybe related, I'm getting
```
'WARNING: drvs not fully operational: 1 sub-bands have not updated in 120 s: drvs7702'
```
I can't quite figure out the numbering scheme for the data recorder

yupinghuang, 2022-12-02:
Here's another one. After stop_dr.
```In [13]: con.status_dr()
Out[13]:
['idle',
 'WARNING: drvs not fully operational: 3 sub-bands error: drvs7802=Only 9 of 10 threads active - MainThread,etcd3_watch_7fc40dfa21d0,Thread-1,Thread-2,CaptureOp,SpectraOp,BaselineOp,StatisticsOp,GlobalLogger; drvs8002=Max. processing time at >125% of gulp (13.440 s = 134.0% ); drvs8201=Max. processing time at >125% of gulp (12.593 s = 125.5% )',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle',
 'idle']```

closed 2023-02-21 by yupinghuang