ovro-lwa/lwa-issues #204: No time synchronization between SNAPS.

rh-codebase, 2022-11-17:
synchronization pulses aren't making it through the Meinberg timing distribution box.

rh-codebase, 2022-11-17:
Meinberg box was power cycled but did not fix the problem. The Meinberg box has 2 LEDs to indicate 'alarm' and 'sync'. Before and after the power cycle the 'sync' LED was on and the 'alarm' led off..
Then, directly connected SNAP1 timing signal to SNAP2 using coax barrel connector then went hunting for an o-scope. This was to allow another test if an o-scope could not be found. A scope was found so the barrel was removed and the TTL timing pulse from SNAP1 was connected to the o-scope. A clean TTL pulse was observed. Connected SNAP1 timing back to the TTL input on the Meinberg distribution box. The 'sync' led turned on. Looked at SNAP2's TTL output from the Meinberg and it also looked clean. Reconnected and looked at the output to the last SNAP and also clean. Reconnected last SNAP output and let Jack know the TTL in/out all looked good. Jack reported timing sync was operational.

ldaddario, 2022-11-17:
Please provide the full model manufacturer name and model number of "the Meinberg box".  Such things need to be documented!

jack-h, 2022-11-17:
Meinberg SDU (1:12 configuration) -- 
[info_sdu.pdf](https://github.com/ovro-lwa/lwa-issues/files/10031596/info_sdu.pdf)

rh-codebase, 2022-11-17:
@jack-h I looked at the TTL output again this morning on the only unused Meinberg port. The pulse width is 1us high-to-low. I noticed what looked like missing pulses. 3 or 4 would show up at a regular cadence then a skip. Are you seeing anything like this on your end?

jack-h, 2022-11-17:
I don't see this. The FPGAs report if the period between successive pulses
is observed to change, and I see none of these events logged over the last
few hours.

closed 2023-02-21 by jack-h