ovro-lwa/lwa-issues #207: power beam data recorder creating bad hdf5 files

caseyjlaw, 2022-11-23:
Ivey and I have occasionally run into a problem when analyzing hdf5 files for power beam data.
We can route x-engine data and start data recorders. They report some packet loss (15.6% in one case, but we were missing an x-engine server). 
We are logging in as user "pipeline" in the "datarecorder" environment. On calim01, the file is at "/data01/beam01/059906_19220895264316a8ec5"
Ivey has the error message when reading the file with h5py. I tried the same thing later, and saw no error while reading on calim01.

iveydavis, 2022-11-23:
For reference, this is the error I got when trying to open the file on calim01:
[image]

I get a different error when I copy to astm and then try to open it there:
[image]

I also noticed the copied file on astm is smaller (file size 1.46GB) than the file on calim (1.47GB). I suspect this might be related to the issue of not being able to open it on calim?

jaycedowell, 2022-11-28:
I think you get this error when the trying to access a HDF5 file that the data recorder still has open internally.

jaycedowell, 2022-11-28:
It should automatically close after some ~10 s grace period at the end of a recording.

caseyjlaw, 2022-11-28:
I see. It is possible we looked at it too soon.
Would it be possible to have the data recorder status reflect this in its status message somehow? For example, "idle" means it has actually completed and closed the file.

jaycedowell, 2022-11-28:
I'm looking at the logs for this recording and it almost looks like there was a restart of the x-engines during the recording.

jaycedowell, 2022-11-28:
Idle should mean that already.

jaycedowell, 2022-11-28:
Ok, it looks like the restart of the data flow stopped the file from getting closed at the end of the recording session.

caseyjlaw, 2022-11-29:
ok, thanks Jayce. Would you like us to install and test this change?

jaycedowell, 2022-11-29:
Everything is already in place.  All that should be required is to restart the services.

closed 2022-12-28 by caseyjlaw